arendst/Tasmota #1673: Lag on Sonoff Tasmota Basic Switch

kckepz, 2018-01-22:
I've flashed my Sonoff Basic switch with the Atom PlatinumIO application, very easy process thank goodness.  I've connected a light rocker switch to my Sonoff Tasmota and find that now and then (quite often) there is a lag when I activate it from my light switch, the button on the Sonoff or from the webpage.  Silly question but, is that the current normal behaviour of the Sonoff with Tasmota firmware?  If it isn't, how do I fix this?  It's really annoying.

Please help.

gajotnt, 2018-01-22:
Is the MQTT server set right? This once happened to me and it was because i dindnt set the MQTT server and it was always trying to connect.

kckepz, 2018-01-22:
Wow!  Was that all it was.... lol.  I'm not using a MQTT server but it was enabled on my Sonoff Tasmota.  I've turned that off and then VOILA!

Thank you @gajotnt

closed 2018-01-22 by kckepz

gajotnt, 2018-01-22:
:)