arendst/Tasmota #1674: TH-10 new board ver 2.1

dendrees, 2018-01-22:
I have received two replacement Th-10. Seems to be there is a new board. Mentioning version 2.1
As well there is no fuse anymore. Anyone tried to flash this one?

andrhrstv, 2018-02-05:
I was able to update mine Sonoff TH16 v2.1 only thru the Arduino IDE and only with those settings:

[image]

use the DOUT Flash mode. 

I was able to flash a custom firmware and also the ESPEasy-r120 

good luck

garymcleanuk, 2018-02-13:
I had issues flashing this last night. However I did a memory upgrade too at the same time. Is it still a case of holding the button while powering up or do we need to ground gpio0 to place into flash mode?

andrhrstv, 2018-02-13:
I'm holding down the button while powering on, works the same.

tomasraf, 2018-03-23:
I've tried to change board_flash_mode to dio and it helped. I'm using Atom + PlatformIO
#board_flash_mode = dout
board_flash_mode = dio