arendst/Tasmota #1675: How to corretly set up PowerOnState, SwitchTopic and SwitchMode with a PIR sensor

krasatos, 2018-01-22:
Hey guys, 
I am coming to you after lots of efforts to figure this out but i seem to be missing something. 
Setup: 

- Sonoff module embeded inside the light fixture.
- Wall switch turns Sonoff module on/off
- Lamp connected to Sonoff output 
- PIR sensor (HC-SR501) connected to Gpio14 and set up as `switch2`
- MQTT enabled and connected to homeassistant built in broker
- Maintopic: `sonoff_stairs`
- Switchtopic `motion/stairs/up`
- Switchmode `1`
- Poweronstate `1`

(keep in mind that i have no neutral cable anywhere, so i am not talking about a 3-way switch configuration)

I have set it up and it works well with my HA configuration, turns on the light whenever motion is detected. 
The problem is when i turn the switch on and off again.

Current behavior:

- I flip the switch off and on
- Sonoff powers up
- Light turns on for 3 seconds
- Light turns back off

Desired Behavior:

- I flip the switch off and on
- Sonoff powers up
- Light turns on for 3 seconds
- Light stays on

Optional, (Even more) Desired Behavior:

- I flip the switch off and on
- Sonoff powers up
- Light turns on for 3 seconds
- Light stays on for a period of time then turns off (either internally or by sending a "fake" motion detected message to my broker, so that the HA turns it off after the set period of time)

I need the switch functionality as my (senior) parents are not used to the motion detection yet.
They automatically hit the switches. So i want this option available too: they hit the switch once, it wont switch on, they hit it twice, it turns and stays on :)

I have a long thread [here ](https://community.home-assistant.io/t/help-with-mqtt-sonoff-with-tasmota-and-poweronstate-command/39222) with all my observations and efforts and a more detailed description of the problem.
I have tried several things as you might read in the thread above, the poweronstate only stays on when i disable the PIR sensor.

Jason2866, 2018-01-22:
In my opinion it is a bad idea to power on and off the Sonoffs.
I think you will get strange behaviours if you will try doing this way.
Always power the Sonoffs and connect the switches to the free GPIO.
If the behavoiur you want isnt possible with this little powerful things do it with a rule in your homeautomation system
Just my 2 Cents
Greets Jason0815

brianbrown1962, 2018-01-22:
I would agree with Jason0815...I've tried a couple of applications in which I tried to power up the sonoff "on demand" but ran into many issues with connectivity to/from wifi and my mqtt server. I too would suggest always having the sonoff powered and do the magic with buttons, pir, and your home automation software.

Jason2866, 2018-01-22:
Hi krasatos,
you can switch individual on and off the sonoffs right?
So i think it should be possible. You have to change a little the wiring or you have to place the sonoffs
in direct near of the switches so that you can wire direct the switch to the GPIO Pin.
Maybe you can upload a wiring scheme from your house installation to see how it is done, and what can be changed...

krasatos, 2018-01-22:
Thing is, that there is no neutral wiring going through my switch. 
The neutral goes directly to the lamp.
The switch just powers the other phase on and off.
(my electrical terminology might be wrong but yiu get the idea)

Is there no way to get the desired functionality software-wise? It seems to me like it should be possible.

Jason2866, 2018-01-22:
Hmm, thats realy minimalistic  wiring...  No poweroutlet nearby?
Sorry fo a tricky software solution here i am not the right guy, and i think it is wasting time.
If i where in your position, i would spent that time in other things maybe install some wires ;-)
A solution (not elegant but will work!). Buy one cheap 433 Mhz receiver. Install this just in place of the sonoff.
Relais contact goes to Gpio. Change the switches against momentary contacts and wire this parallel to the on button of a little 433 Mhz Transmitter, that you can place behind. With this you can get so many switches you want... Pir sensors and other magic things will work 100%. Other advantage if the homeautomation is down the solution works as a "normal" switch. 
On Aliexpress you find this things for a few Euros.

krasatos, 2018-01-23:
Thanks for using the term "minimalistic".
unfortunately, if there is no power outlet under the switches, there is no neutral wire in the switch.

I keep thinking that this is a software issue though, **why wont a sonoff just turn on and stay on (after booting) if a sensor is connected to it?**

davidelang, 2018-01-23:
enable detailed logging and see what it reports is happening.

b1g1an, 2018-01-23:
You're not wiring it correctly for what you want. All wiring is still at the light fitting though, no need for anything extra.

You shouldn't use the existing light switch to cut mains power to the Sonoff, you should be using it attached to the GPIO and set to Switch 1 to trigger the relay which would then give you normal on off light at the switch which would be reflected in HA so everything is in sync. Then connect the PIR sensor to another GPIO terminal and set to Switch 2, lastly set a spare GPIO to Relay 2. Then in HA a topic ending Power1 will directly control the relay in the Sonoff and the change in state of Power2 can be used to trigger Power1.

There are several ways to achieve the same thing using switchtopic instead but I like the above as it means should your wifi or HA go down you will have normal control of the light from the existing light switch.

krasatos, 2018-01-23:
Hey @davidelang thanks for checking in, i enabled full web logging:
[image]

Here is the output on boot up after manual off and on:

```
00:00:00 CFG: Loaded from flash at FA, Count 818
00:00:00 APP: Boot Count 106
00:00:00 Project sonoff Sonoff Stairs Up (Topic sonoff_stairs, Fallback sonoff_stairs_up, GroupTopic sonoffs) Version 5.9.1
00:00:00 CFG: Saved to flash at F9, Count 819, Bytes 1648
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 XXXX in mode 11N as sonoff_stairs-4940...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 RSL: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
00:00:02 RSL: stat/sonoff_stairs/POWER = OFF
00:00:03 CFG: Saved to flash at F8, Count 820, Bytes 1648
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff_stairs-4940.local with IP address XXXXX
00:00:04 APP: (UTC) Tue Jan 23 12:04:01 2018
00:00:04 APP: (DST) Sun Mar 25 02:00:00 2018
00:00:04 APP: (STD) Sun Oct 28 03:00:00 2018
14:04:02 MQT: Attempting connection...
14:04:04 MQT: Connected
14:04:04 MQT: tele/sonoff_stairs/LWT = Online (retained)
14:04:04 MQT: cmnd/sonoff_stairs/POWER = 
14:04:04 MQT: Subscribe to cmnd/sonoff_stairs/#
14:04:04 MQT: Subscribe to cmnd/sonoffs/#
14:04:04 MQT: Subscribe to cmnd/sonoff_stairs_up/#
14:04:04 MQT: tele/sonoff_stairs/INFO1 = {"Module":"Sonoff Basic", "Version":"5.9.1", "FallbackTopic":"sonoff_stairs_up", "GroupTopic":"sonoffs"}
14:04:04 MQT: tele/sonoff_stairs/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff_stairs-4940", "IPAddress":"XXXX"}
14:04:04 MQT: tele/sonoff_stairs/INFO3 = {"RestartReason":"Power on"}
14:04:05 MQT: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
14:04:05 MQT: stat/sonoff_stairs/POWER = OFF14:01:29 WIF: Connected
14:01:41 HTP: Main Menu
14:01:47 HTP: Configuration
14:01:50 HTP: Configure Logging
14:01:51 WIF: Checking connection...
14:01:51 WIF: Connected
14:02:00 MQT: tele/sonoff_stairs/UPTIME = {"Time":"2018-01-23T14:02:00", "Uptime":17}
14:02:11 WIF: Checking connection...
14:02:11 WIF: Connected
14:02:31 WIF: Checking connection...
14:02:31 WIF: Connected
14:02:41 MQT: tele/sonoff_stairs/STATE = {"Time":"2018-01-23T14:02:41", "Uptime":17, "Vcc":3.198, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXXX", "RSSI":80, "APMac":"XXXXX"}}
14:02:41 MQT: tele/sonoff_stairs/SENSOR = {"Time":"2018-01-23T14:02:41", "Switch1":"OFF"}
14:02:51 WIF: Checking connection...
14:02:51 WIF: Connected
14:03:09 HTP: Configuration
14:03:12 WIF: Checking connection...
14:03:12 WIF: Connected
14:03:14 HTP: Main Menu
14:03:16 HTP: Console
14:04:04 MQT: tele/sonoff_stairs/INFO3 = {"RestartReason":"Power on"}
14:04:05 MQT: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
14:04:05 MQT: stat/sonoff_stairs/POWER = OFF
14:04:09 MQT: cmnd/motion/stairs/up/POWER1 = ON
14:04:10 HTP: Console
14:04:13 MQT: cmnd/motion/stairs/up/POWER1 = OFF
14:04:14 MQT: tele/sonoff_stairs/STATE = {"Time":"2018-01-23T14:04:14", "Uptime":0, "Vcc":3.198, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXXX", "RSSI":76, "APMac":"XXXXX"}}
14:04:14 MQT: tele/sonoff_stairs/SENSOR = {"Time":"2018-01-23T14:04:14", "Switch1":"OFF"}
14:04:26 WIF: Checking connection...
14:04:26 WIF: Connected
14:04:28 MQT: cmnd/motion/stairs/up/POWER1 = ON
14:04:31 MQT: cmnd/motion/stairs/up/POWER1 = OFF
```

krasatos, 2018-01-23:
@b1g1an , I fail to see how this is a wiring issue, or how i can wire it differently.
The way my house is wired is: 
`Switchboard -> single power cable -> Switch -> single power cable -> Light Fixture.`
then
`Switchboard -> single neutral cable (not from the same route as above) -> Light fixture `

I dont see how i should connect 220v cable coming from the switch to my gpio.
And i dont see a way to get new cables through unless i start breaking the wall :)

Unless i am missing something :confused:

b1g1an, 2018-01-23:
Yep, you're missing something as that's how I assumed it was wired :) 

You remove the switch from the power circuit completely at the light and put the power straight into the Sonoff, it's then live all the time. Next connect the two wires from the switch to a GPIO connection, usually 14, and ground on the Sonoff. Set it up as has been discussed ad infinitum. Then add the PIR as I said before.

I posted a diagram of how to connect up the switch and Sonoff at the light fitting in a discussion on the HA board about 2 way circuits recently.

krasatos, 2018-01-23:
@b1g1an The sonoff is positioned at the switch? or at the fixture?

b1g1an, 2018-01-23:
Sonoff at the fixture.

Looking at your description again, you're making me wonder if i'm confused about your wiring. 

From what i've seen there are only two ways of switching a light, firstly, and normal (until recently) for UK is all wiring goes to light fitting and then just a live feed goes to the switch and then back to the fitting, which is what i assumed you had. Second way is the normal US way where wiring goes to the switch and then just switched power is fed to the light, this is also used with UK outside lights.

Apologies if have you got something very different again.

krasatos, 2018-01-23:
I think we sorted it out, :+1: 
That's not how i'm wired.
This requires 4 wires at either place (the light or the switch). Two live ones (live+neutral) and two to connect either the light or the switch, depending on the examples you described. In that cases what you described is possible.

In my case, and since the switch is far away from the light, think of it as one single cable, running from the power, to the switch, then to the fixture, then to neutral in a straight line.

Jason2866, 2018-01-23:
Hi krastos,
i tried your setup with a module that i have planned for a "future project"
This modul:
https://ucexperiment.wordpress.com/2016/12/18/yunshan-esp8266-250v-15a-acdc-network-wifi-relay-module/
With: Switchmode 1 and Poweronstate 1 i get the results you want. For the switch i used the port connected to the terminal (via optocoupler). The only thing that i have left out is the PIR Sensor.
If the PIR makes the problem, seperate that pice with his own ESP Modul...
Here is my log:
tele/sonoff-06A826/STATE = {"Time":"2018-01-23T15:04:04","Uptime":1,"Vcc":3.213,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":82,"APMac":"00:A0:57:2A:BD:19"}}
15:09:08 MQT: tele/sonoff-06A826/STATE = {"Time":"2018-01-23T15:09:08","Uptime":1,"Vcc":3.215,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":82,"APMac":"00:A0:57:2A:BD:19"}}
15:11:34 APP: Button 1 multi-press 1
15:11:34 MQT: stat/sonoff-06A826/RESULT = {"POWER":"ON"}
15:11:34 MQT: stat/sonoff-06A826/POWER = ON
15:11:34 CFG: Saved to flash at F9, Count 94, Bytes 1648
after powering on:
00:00:00 CFG: Loaded from flash at F9, Count 94
00:00:00 APP: Boot Count 24
00:00:00 Project sonoff Relais (Topic sonoff-06A826, Fallback sonoff-06A826, GroupTopic sonoffs) Version 5.11.1
00:00:00 CFG: Saved to flash at F8, Count 95, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 Jason_Home_WLAN in mode 11N as sonoff-06A826-2086...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff-06A826-2086.local with IP address 192.168.2.132
00:00:04 UPP: Multicast (re)joined
00:00:05 UPP: Multicast disabled
00:00:05 HTP: Console
00:00:06 UPP: Multicast (re)joined
00:00:06 APP: (UTC) Tue Jan 23 14:12:45 2018
00:00:06 APP: (DST) Sun Mar 25 02:00:00 2018
00:00:06 APP: (STD) Sun Oct 28 03:00:00 2018
15:12:46 UPP: Multicast disabled
15:12:46 MQT: Attempting connection...
15:12:46 MQT: Connected
15:12:46 MQT: tele/sonoff-06A826/LWT = Online (retained)
15:12:46 MQT: cmnd/sonoff-06A826/POWER = 
15:12:46 MQT: Subscribe to cmnd/sonoff-06A826/#
15:12:46 MQT: Subscribe to cmnd/sonoffs/#
15:12:46 MQT: Subscribe to cmnd/sonoff-06A826/#
15:12:46 MQT: tele/sonoff-06A826/INFO1 = {"Module":"Yunshan Relay","Version":"5.11.1","FallbackTopic":"sonoff-06A826","GroupTopic":"sonoffs"}
15:12:46 MQT: tele/sonoff-06A826/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-06A826-2086","IPAddress":"192.168.2.132"}
15:12:46 MQT: tele/sonoff-06A826/INFO3 = {"RestartReason":"Power on"}
15:12:47 UPP: Multicast (re)joined
15:12:47 MQT: stat/sonoff-06A826/RESULT = {"POWER":"ON"}
15:12:47 MQT: stat/sonoff-06A826/POWER = ON
15:12:54 MQT: tele/sonoff-06A826/STATE = {"Time":"2018-01-23T15:12:54","Uptime":0,"Vcc":3.222,"POWER":"ON","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":70,"APMac":"00:A0:57:2A:BD:19"}}

and the switch log (pressed 2 times off, on)

15:14:15 APP: Button 1 multi-press 1
15:14:15 MQT: stat/sonoff-06A826/RESULT = {"POWER":"OFF"}
15:14:15 MQT: stat/sonoff-06A826/POWER = OFF
15:14:16 CFG: Saved to flash at F7, Count 96, Bytes 1648
15:14:48 APP: Button 1 multi-press 1
15:14:49 MQT: stat/sonoff-06A826/RESULT = {"POWER":"ON"}
15:14:49 MQT: stat/sonoff-06A826/POWER = ON
15:14:49 CFG: Saved to flash at F6, Count 97, Bytes 1648

davidelang, 2018-01-24:
At 14:04:09 it received the command to power on
at 14:04:13 it received the command to power off

was this 'more debugging'? if not, switch to that and see if it tells you where 
those commands come from

On Tue, 23 Jan 2018, krasatos wrote:

> Date: Tue, 23 Jan 2018 04:07:16 -0800
> From: krasatos <notifications@github.com>
> Reply-To: arendst/Sonoff-Tasmota
>     <reply+0021b24f9269da0ce1715d923b574f7362b9b2af253df2b892cf00000001167ee77
>     492a169ce1150bf56@reply.github.com>
> To: arendst/Sonoff-Tasmota <Sonoff-Tasmota@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Mention <mention@noreply.github.com>
> Subject: Re: [arendst/Sonoff-Tasmota] How to corretly set up PowerOnState,
>     SwitchTopic and SwitchMode with a PIR sensor (#1675)
> 
> Hey @davidelang thanks for checking in, i enabled full web logging:
> [image]
>
> Here is the output on boot up after manual off and on:
>
> ```
> 00:00:00 CFG: Loaded from flash at FA, Count 818
> 00:00:00 APP: Boot Count 106
> 00:00:00 Project sonoff Sonoff Stairs Up (Topic sonoff_stairs, Fallback sonoff_stairs_up, GroupTopic sonoffs) Version 5.9.1
> 00:00:00 CFG: Saved to flash at F9, Count 819, Bytes 1648
> 00:00:00 WIF: Checking connection...
> 00:00:00 WIF: Attempting connection...
> 00:00:00 WIF: Patch issue 2186
> 00:00:00 WIF: Connecting to AP1 XXXX in mode 11N as sonoff_stairs-4940...
> 00:00:01 WIF: Checking connection...
> 00:00:01 WIF: Attempting connection...
> 00:00:02 WIF: Checking connection...
> 00:00:02 WIF: Attempting connection...
> 00:00:02 RSL: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
> 00:00:02 RSL: stat/sonoff_stairs/POWER = OFF
> 00:00:03 CFG: Saved to flash at F8, Count 820, Bytes 1648
> 00:00:03 WIF: Checking connection...
> 00:00:03 WIF: Connected
> 00:00:03 DNS: Initialized
> 00:00:03 HTP: Web server active on sonoff_stairs-4940.local with IP address XXXXX
> 00:00:04 APP: (UTC) Tue Jan 23 12:04:01 2018
> 00:00:04 APP: (DST) Sun Mar 25 02:00:00 2018
> 00:00:04 APP: (STD) Sun Oct 28 03:00:00 2018
> 14:04:02 MQT: Attempting connection...
> 14:04:04 MQT: Connected
> 14:04:04 MQT: tele/sonoff_stairs/LWT = Online (retained)
> 14:04:04 MQT: cmnd/sonoff_stairs/POWER =
> 14:04:04 MQT: Subscribe to cmnd/sonoff_stairs/#
> 14:04:04 MQT: Subscribe to cmnd/sonoffs/#
> 14:04:04 MQT: Subscribe to cmnd/sonoff_stairs_up/#
> 14:04:04 MQT: tele/sonoff_stairs/INFO1 = {"Module":"Sonoff Basic", "Version":"5.9.1", "FallbackTopic":"sonoff_stairs_up", "GroupTopic":"sonoffs"}
> 14:04:04 MQT: tele/sonoff_stairs/INFO2 = {"WebServerMode":"Admin", "Hostname":"sonoff_stairs-4940", "IPAddress":"XXXX"}
> 14:04:04 MQT: tele/sonoff_stairs/INFO3 = {"RestartReason":"Power on"}
> 14:04:05 MQT: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
> 14:04:05 MQT: stat/sonoff_stairs/POWER = OFF14:01:29 WIF: Connected
> 14:01:41 HTP: Main Menu
> 14:01:47 HTP: Configuration
> 14:01:50 HTP: Configure Logging
> 14:01:51 WIF: Checking connection...
> 14:01:51 WIF: Connected
> 14:02:00 MQT: tele/sonoff_stairs/UPTIME = {"Time":"2018-01-23T14:02:00", "Uptime":17}
> 14:02:11 WIF: Checking connection...
> 14:02:11 WIF: Connected
> 14:02:31 WIF: Checking connection...
> 14:02:31 WIF: Connected
> 14:02:41 MQT: tele/sonoff_stairs/STATE = {"Time":"2018-01-23T14:02:41", "Uptime":17, "Vcc":3.198, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXXX", "RSSI":80, "APMac":"XXXXX"}}
> 14:02:41 MQT: tele/sonoff_stairs/SENSOR = {"Time":"2018-01-23T14:02:41", "Switch1":"OFF"}
> 14:02:51 WIF: Checking connection...
> 14:02:51 WIF: Connected
> 14:03:09 HTP: Configuration
> 14:03:12 WIF: Checking connection...
> 14:03:12 WIF: Connected
> 14:03:14 HTP: Main Menu
> 14:03:16 HTP: Console
> 14:04:04 MQT: tele/sonoff_stairs/INFO3 = {"RestartReason":"Power on"}
> 14:04:05 MQT: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
> 14:04:05 MQT: stat/sonoff_stairs/POWER = OFF
> 14:04:09 MQT: cmnd/motion/stairs/up/POWER1 = ON
> 14:04:10 HTP: Console
> 14:04:13 MQT: cmnd/motion/stairs/up/POWER1 = OFF
> 14:04:14 MQT: tele/sonoff_stairs/STATE = {"Time":"2018-01-23T14:04:14", "Uptime":0, "Vcc":3.198, "POWER":"OFF", "Wifi":{"AP":1, "SSId":"XXXXX", "RSSI":76, "APMac":"XXXXX"}}
> 14:04:14 MQT: tele/sonoff_stairs/SENSOR = {"Time":"2018-01-23T14:04:14", "Switch1":"OFF"}
> 14:04:26 WIF: Checking connection...
> 14:04:26 WIF: Connected
> 14:04:28 MQT: cmnd/motion/stairs/up/POWER1 = ON
> 14:04:31 MQT: cmnd/motion/stairs/up/POWER1 = OFF
> ```
>
>
>

krasatos, 2018-02-02:
@davidelang Hey, sorry for taking so long to reply, i was out of the country.

Yes, the log are with more debugging enabled.
I don't see how to understand where the commands came from.
I am attaching another log (with "more debugging" enabled)
The broker was offline at that point (i powered off my RPi just before powering off and on the device)
While the broker is offline, it turns on and off **twice**, first time for ~3 sec, then for about ~2
I'm more confused now :confused: 

```
00:00:00 CFG: Loaded from flash at F7, Count 1197
00:00:00 APP: Boot Count 122
00:00:00 Project sonoff Sonoff Stairs Up (Topic sonoff_stairs, Fallback sonoff_stairs_up, GroupTopic sonoffs) Version 5.9.1
00:00:00 CFG: Saved to flash at F6, Count 1198, Bytes 1648
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 XXXXX in mode 11N as sonoff_stairs-4940...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:02 RSL: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
00:00:02 RSL: stat/sonoff_stairs/POWER = OFF
00:00:03 CFG: Saved to flash at F5, Count 1199, Bytes 1648
00:00:03 WIF: Checking connection...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff_stairs-4940.local with IP address 10.0.0.223
00:00:03 UPP: Multicast (re)joined
00:00:04 APP: (UTC) Fri Feb 02 14:09:13 2018
00:00:04 APP: (DST) Sun Mar 25 02:00:00 2018
00:00:04 APP: (STD) Sun Oct 28 03:00:00 2018
16:09:14 UPP: Multicast disabled
16:09:14 MQT: Attempting connection...
16:09:20 MQT: Connect failed to XXXXXXX:1883, rc -2. Retry in 10 sec
16:09:20 RSL: stat/sonoff_stairs/RESULT = {"POWER":"ON"}
16:09:20 RSL: stat/sonoff_stairs/POWER = ON
16:09:20 CFG: Saved to flash at F4, Count 1200, Bytes 1648
16:09:21 UPP: Multicast (re)joined
16:09:22 RSL: stat/sonoff_stairs/RESULT = {"POWER":"OFF"}
16:09:22 RSL: stat/sonoff_stairs/POWER = OFF
16:09:23 CFG: Saved to flash at FB, Count 1201, Bytes 1648
16:09:32 UPP: Multicast disabled
16:09:32 MQT: Attempting connection...
16:09:38 MQT: Connect failed to XXXXXX:1883, rc -2. Retry in 10 sec
16:09:39 UPP: Multicast (re)joined
```

Jason2866, 2018-02-02:
Hi,
for testing i would disconnect the pir and all other things if there are any......
I had a pir that makes "funny" things. A pulldown resistor was needed to get it working right

krasatos, 2018-02-02:
Hey, thanks for the input @jason2866.

Of I disable the sensors from the ui, the poweronstate does work. 

I don't think it's defective sensors causing the problem though, as I have two lights, both setup the same way and both have the same operation. They switch off after 3 seconds of boot up.

It must be something I have messed up in mqtt...

Jason2866, 2018-02-02:
Hi,
is it working if the pi is powerd off?
Have you already traced the mqqt traffic?
This command helps (you have to replace the sonoff topic with yours):
$ mosquitto_sub -h localhost -t "+/sonoff-E8A6E4/#" -v

krasatos, 2018-02-03:
And this brings us to a whole new problem i encountered while trying to figure out how to run these commands [here](https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration). 

As i have Hassio i cannot run the command on the RPi itself.

I have set up a VM on my win pc with ubuntu in order to troubleshoot but, although all my network settings seem to be ok (i can ping things around) i cannot sunscribe to the broker. 

Getting a spam of `Client mosqsub/22618-username- sending CONNECT`

Further info on this issue [here](https://community.home-assistant.io/t/connect-to-embedded-mqtt-broker-via-ubuntu-on-local-win7-vm/41788/13).

Jason2866, 2018-02-11:
Hi,
i tried this setup now with a pir connected too. Same behaviour as your setup. BUT this i correct because the PIR module that i have toggles the GPIO Pin 14 during power up! Maybe you can solve it, with changing the time adjust on the PIR module? If this doesnt help, you have to connect the PIR to his own modul.