arendst/Tasmota #1677: POW Today/Yesterday energy use not resetting automatically

jjlawren, 2018-01-22:
Since updating to 5.11.X, the daily energy use reporting is not resetting properly. For example, I used 0.500kWh yesterday and none today, but both yesterday and today report as 0.500kWh.

When issuing the EnergyReset command to query the energy totals, both are displayed correctly (ie, today is zero).

Console output:
```
13:34:23 MQT: tele/sonoff_POW_dryer/SENSOR = {"Time":"2018-01-22T13:34:23","ENERGY":{"Total":46.843,"Yesterday":0.500,"Today":0.500,"Period":0,"Power":0,"Factor":0.00,"Voltage":118,"Current":0.000}}
13:39:06 MQT: stat/sonoff_POW_dryer/RESULT = {"EnergyReset":{"Total":46.843,"Yesterday":0.500,"Today":0.000}}
13:39:23 MQT: tele/sonoff_POW_dryer/STATE = {"Time":"2018-01-22T13:39:23","Uptime":141,"Vcc":3.536,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxx","RSSI":68,"APMac":"xxxx"}}
13:39:23 MQT: tele/sonoff_POW_dryer/SENSOR = {"Time":"2018-01-22T13:39:23","ENERGY":{"Total":46.843,"Yesterday":0.500,"Today":0.500,"Period":0,"Power":0,"Factor":0.00,"Voltage":118,"Current":0.000}}
13:40:53 MQT: stat/sonoff_POW_dryer/RESULT = {"EnergyReset":{"Total":46.843,"Yesterday":0.500,"Today":0.000}}
```

jjlawren, 2018-01-23:
Looks like any power usage immediately resets today:
```
20:19:24 MQT: tele/sonoff_POW_dryer/SENSOR = {"Time":"2018-01-22T20:19:24","ENERGY":{"Total":46.843,"Yesterday":0.500,"Today":0.500,"Period":0,"Power":0,"Factor":0.00,"Voltage":118,"Current":0.000}}
20:22:55 MQT: tele/sonoff_POW_dryer/MARGINS = {"PowerLow":"OFF","PowerHigh":"ON"}
20:22:55 MQT: tele/sonoff_POW_dryer/ENERGY = {"Time":"2018-01-22T20:22:55","ENERGY":{"Total":46.843,"Yesterday":0.500,"Today":0.500,"Power":11,"Factor":0.00,"Voltage":118,"Current":0.000}
20:23:26 MQT: tele/sonoff_POW_dryer/MARGINS = {"PowerLow":"ON","PowerHigh":"OFF"}
20:23:26 MQT: tele/sonoff_POW_dryer/ENERGY = {"Time":"2018-01-22T20:23:26","ENERGY":{"Total":46.843,"Yesterday":0.500,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":118,"Current":0.000}
```

jjlawren, 2018-01-23:
It appears that 'Yesterday' resets properly at midnight and the problem is specific to 'Today' only resetting when power is consumed in a "new" day.

arendst, 2018-01-24:
Will investigate

jjlawren, 2018-01-25:
Seems closely related to https://github.com/arendst/Sonoff-Tasmota/issues/1648.

Frogmore42, 2018-02-06:
I am seeing the same behavior since updating to 5.11.1h.  It resets correctly if there is power usage/relay is on (not sure which it really requires, since both happen at the same time).

arendst, 2018-02-06:
Already fixed it. Will need to push to github today as an update of 5.11.1i.

closed 2018-02-09 by arendst