arendst/Tasmota #1680: Buttontopic, grouptopic for button2 sonoff T1 2CH

h3lmutek, 2018-01-22:
Hi, how i configure buttontopic and grouptopic for button2, button3 etc. on sonoff dual, t1 2CH, pro 4CH..thanks for much

h3lmutek, 2018-02-28:
thanks for answer! change firmware to espurna - great grouptopic for button2,3 etc

h3lmutek, 2018-05-28:
on tasmota no, i changed firmware to espurna.

ascillato2, 2018-08-23:
Hi,

It is fixed in last dev version. Please, try it. Thanks

closed 2018-08-23 by ascillato2

andreas6, 2018-08-25:
Tested today with the latest compiled version. Didn't work for me. Only 1st channel responds to group topic commands.
Am I missing something?

andreas6, 2018-08-25:
To clarify, I compiled myself the latest development version.

arendst, 2018-08-25:
What you expect from grouptopic is not how it is conceived.

Grouptopic addresses a group of devices.

So on a 4CH you still have to provide the subtopic power2, power3 etc.

andreas6, 2018-08-25:
arendst, still didn't get it.

I have the " Sonoff T1 2CH EU".
Have set grouptopic to "lights" and want to turn off all lights before leave home.
Second channel of the T1 2CH doesn't respond to grouptopic "lights".
If I can't do that then how it is fixed as mentioned above?

If the fix is the referred "subtopic power2" please provide more info because can't figure out how and where to use it. 

Thank you.

andreas6, 2018-08-25:
This command works:         cmnd/lights/power2 off
But this is not what I want.
I want to press and hold the button near my entrance door and turn off all the house lights.
I used ButtonTopic=lights for that.
How to do that for channel 2. Do I need to add rules?

arendst, 2018-08-25:
I try to explain again...

Grouptopic addresses several devices listening to the grouptopic. It does not know about subtopics like POWER2 or POWER3.

If you want to turn of all power from your single 4CH you will need to make a rule for that. Once done you can use grouptopic to execute that rule which in turn will do what you want.

andreas6, 2018-08-27:
I read the Rule syntax and examples.
Can't find how to trigger a rule from the MQTT command:
 cmnd/lights/power off

This MQTT command is published from a specific Sonoff to every Sonoff in house. I don't want to use openHAB rules for that.
If there is a way please give me a hint. I don't want the final solution. I want to use my brain.