arendst/Tasmota #1681: webserver not responding

ionciubotaru, 2018-01-23:
I tried to compile Tasmota latest version, 5.11.1d with arduino 1.8.5 and esp library 2.4.0
Everything works fine, except the web server - it doesn't respond

```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0CD6BD, GroupTopic sonoffs) Version 5.11.1d
00:00:00 WIF: Connecting to AP1 Clicknet-BB4A in mode 11N as sonoff-5821...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff-5821.local with IP address 192.168.1.97
11:05:20 MQT: Attempting connection...
11:05:20 MQT: Connected
11:05:20 MQT: tele/sonoff/LWT = Online (retained)
11:05:20 MQT: cmnd/sonoff/POWER = 
11:05:20 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1d","FallbackTopic":"DVES_0CD6BD","GroupTopic":"sonoffs"}
11:05:20 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5821","IPAddress":"192.168.1.97"}
11:05:20 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
11:05:20 MQT: homeassistant/switch/sonoff/config =  (retained)
11:05:21 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
11:05:21 MQT: stat/sonoff/POWER = OFF
11:11:45 CMD: status 0
11:11:45 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
11:11:45 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":8,"SaveCount":19,"SaveAddress":"F9000"}}
11:11:45 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.11.1d","BuildDateTime":"2018-01-23T11:56:05","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
11:11:45 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Test","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
11:11:45 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":518,"Free":484,"Heap":10,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:11:45 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5821","IPAddress":"192.168.1.97","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.253","Mac":"5C:CF:7F:0C:D6:BD","Webserver":2,"WifiConfig":3}}
11:11:45 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0CD6BD","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:11:45 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Jan 23 10:11:45 2018","Local":"Tue Jan 23 11:11:45 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
11:11:45 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-01-23T11:11:45"}}
11:11:45 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-01-23T11:11:45","Uptime":0,"Vcc":3.171,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Test","RSSI":86,"APMac":"BC:C0:1F:24:C9:30"}}}

```

ascillato, 2018-01-23:
have you done a PING command from a PC to your Sonoff 192.168.1.97?

ionciubotaru, 2018-01-23:
YES, ping is working fine, sonoff connects to MQTT, get the time from internet, everything except web interface. The same script compiled with 2.3.0 works fine. I'm not sure if it's a Tasmota or esp library problem.

ascillato, 2018-01-23:
I compile Tasmota v5.11.1d with arduino 1.8.2 and esp library 2.4.0 and works fine. Did you make any changes on any file before compiling?

ascillato, 2018-01-23:
May be an upload error. Have you the flash configuration on arduino the same as the wiki?

ionciubotaru, 2018-01-23:
NO, I used the standard version, downloaded from tasmota's github

arendst, 2018-01-24:
By not responding you mean nothing happens or do you receive a blank page? If blank try to retrieve the code of the page from your browser.

I do see a very low heap (10) which shouldn't be there just after a reboot ... This amounts to the lack of a functional webserver as currently the webpage is built in RAM before it is send. I'm working on chunked HTTP traffic for the larger pages but at least the main menu page should pop up anyway.

ionciubotaru, 2018-01-24:
In Chrome debugger when I try to load the page I receive: `STATUS 200` and in `TIME` column `PENDING`
The same software compiled with 2.3.0 works fine. I upload the software via serial and via web interface (available in 2.3.0 compiled version) to avoid any flash programming error.

Tasmota is standard version without any changes.

arendst, 2018-01-24:
In Arduino IDE - Tools (v2.4.0) what LwIP Variant do you use?

ionciubotaru, 2018-01-24:
Windows 7
Arduino IDE 1.8.5 Windows ZIP file for non admin install
ESP 8266 2.4.0 installed via Boards Manager from http://arduino.esp8266.com/stable/package_esp8266com_index.json 
After instalation I replaced boards.txt and platform.txt with the files provided by Tasmota 
Before compiling I selected ESP8285 and  "Flash Size 1Mb (no spiffs)"

ionciubotaru, 2018-01-24:
Sorry I don't know howto check the ` LwIP Variant`

arendst, 2018-01-24:
Although it does not amount to your problem you should not need to select ESP8285. See picture below about correct settings:

[image]

As you can see there is a new entry for LwIP. The selected option should work. Option V2 Prebuilt (MSS 536) will initiate WatchDog. Pls verify your settings as these settings work fine with me.

ionciubotaru, 2018-01-25:
I tried the last version 5.11.1e, I updated  `boards.txt` and `platform.txt` and problem still persist. 
Both 8266 and 8285 has the same issue. The `LwIP Variant` is not present in Tools menu.

[image]

I disabled as many additional options I can, just to maximize free memory:
```
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0CD6BD, GroupTopic sonoffs) Version 5.11.1e
00:00:00 WIF: Connecting to AP1 Test in mode 11N as sonoff-5821...
00:00:04 WIF: Connected
00:00:04 HTP: Web server active on sonoff-5821 with IP address 192.168.1.97
11:59:18 MQT: Attempting connection...
11:59:18 MQT: Connected
11:59:18 MQT: tele/sonoff/LWT = Online (retained)
11:59:18 MQT: cmnd/sonoff/POWER = 
11:59:18 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.11.1e","FallbackTopic":"DVES_0CD6BD","GroupTopic":"sonoffs"}
11:59:18 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5821","IPAddress":"192.168.1.97"}
11:59:18 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
11:59:19 MQT: stat/sonoff/RESULT = {"POWER":"OFF"}
11:59:19 MQT: stat/sonoff/POWER = OFF (retained)
11:59:31 CMD: status 0
11:59:31 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":0,"SaveState":0,"ButtonRetain":0,"PowerRetain":1}}
11:59:31 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.1.253:80/s.ino.bin","Uptime":0,"Sleep":0,"BootCount":4,"SaveCount":6,"SaveAddress":"FB000"}}
11:59:31 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.11.1e","BuildDateTime":"2018-01-25T11:05:22","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
11:59:31 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Test","SSId2":"Test2","TelePeriod":300,"SetOption":"00000028"}}
11:59:31 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":436,"Free":564,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
11:59:31 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5821","IPAddress":"192.168.1.97","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.253","Mac":"5C:CF:7F:0C:D6:BD","Webserver":2,"WifiConfig":4}}
11:59:31 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.253","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0CD6BD","MqttUser":"DVES_USER","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:59:31 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 25 09:59:31 2018","Local":"Thu Jan 25 11:59:31 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99}}
11:59:31 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-01-25T11:59:31"}}
11:59:31 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-01-25T11:59:31","Uptime":0,"Vcc":3.061,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Test","RSSI":56,"APMac":"BC:C0:0F:64:C9:30"}}}
```
[image]

arendst, 2018-01-25:
Strange mine work fine.

Are you sure the IP address of your gateway/router is 192.168.1.1 while your DNS is 192.168.2.253? Mine are different too as I run dnsmasq but just for the fun...

ionciubotaru, 2018-01-25:
My DNS and DHCP and WEB server si a raspberry pi with 192.168.1.253. I cannot trust the router's DNS and DHCP because I have more than 100 Sonoffs in house + laptops + phones.
This configuration works fine for more than 5 months.

arendst, 2018-01-25:
Just uploaded the correct boards.txt file. Pls try again.

ionciubotaru, 2018-01-25:
I updated `boards.txt`, now LwIP selection appears in tools menu, I set it to `v 1.4` , but the problem still persist.
I uploaded `HelloServer` sketch to sonoff (just to check) and it works fine - of course in this case the returned web page is very small.

arendst, 2018-01-25:
Strange. I have it (5.11.1e) compiled with platformio running on 30+ devices without a hitch. 

Will try with Arduino IDE 1.8.5 and 2.4.0 on a sonoff basic and let you know my results.

ionciubotaru, 2018-01-25:
I succeeded compiling with `V2 prebuilt (MSS=536)` Now Information page show `Core/SDK Version | 2_4_0/2.1.0(deb1901)`

I clean the browser cache several times, even use it in incognito mode and the web server is not available with version 1.4

arendst, 2018-01-25:
V2 is a bit challenging.... Many people have problems with it resulting in WDT's just after Wifi connection. General advice is not to use V2 in 2.4.0 (https://github.com/esp8266/Arduino/issues/4078)

arnold-b, 2018-01-27:
Hello, 
i have the same problem. But only on 3 Windows machine. Arb Cache is ok.  
Problem exist in Firefox, chrome or IE. 

On a windows 2013 VM it works. 

In My Config i can´t setup lwIP on the Board.

thanks Arnold

wayne47git, 2018-01-30:
Similar problem here. Just flashed my first unit. Running lynx works. Running Chrome from desktop (wired), Notebook (wireless) and Android tablet fails. Running from an Amazon Fire works (silk). Firefox fails similar to Chrome. ping works everywhere.  Here are some screenshots that might help.

Load is sonoff, 5.11.1 (latest), installed over sonoff_minimal

 Wireshark showing the get:
[image]

The initial response from the Sonoff
[image]

Second and final packet from the Sonoff
[image]

tobox, 2018-02-08:
Not sure, if it is related, but after I recompiled the firmware without emulation support, the problem disappeared (I have various UPNP-devices in my LAN, maybe that overwhelmed the ESP).

mattpurnell, 2018-02-22:
I have the same issue - pings OK, and the authentication is reached - then nothing. Tried multiple browsers, including an android

pmandryk, 2018-03-15:
Same here. I've got PRTG on my sonoffs and it looks like they are losing connectivity to the wifi. They come back online themselves, then drop off again with no intervention.

I'm looking in to the wifi right now.

SecurityAddicted, 2018-03-22:
Same issue here with Tasmota 5.12.0 - SDK 2.4.1 - lwIP v1.4. Changing to v2 Low Memory (mss536) seems to solve the problem.

SecurityAddicted, 2018-03-22:
One thing I noticed on lwIP v1.4 is, if you let a ping go, when you navigate to the webserver the ping stops getting back.

ionciubotaru, 2018-03-22:
In my case, compiling with 2.4.1 lwIP v1.4  the web interface works fine.
```
{"StatusFWR":{"Version":"5.12.0e","BuildDateTime":"2018-03-16T17:51:47","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
```

ascillato, 2018-06-04:
Hi @ionciubotaru

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-05 by ionciubotaru