arendst/Tasmota #1690: WS2811 RGB LED Stripe mixed up colors

roesl, 2018-01-24:
Hallo,

i tried to connect a WS2811 LED stripe 12 V and used WS2812 in Tasmota - but the stripes seems to be RBG instead of RGB with B and G mixed up
Is it possible to add configuration for WS2811 stripes

Regards
Georg

reloxx13, 2018-01-24:
just solder it right.

roesl, 2018-01-24:
ist a 12 V led stripe with controller WS2811  soldered at the stripe so I cant change the colours

legionGer, 2018-01-27:
I think you can just cut the plug off and resolder it.

roesl, 2018-01-27:
Ist a LED stripe - so its necessary to change the RGB org GRB within the Software to RBG but there ist only the Option GRB or RGB but not RBG in the config file for the 2812 ( 2811)

lobradov, 2018-01-27:
You might want to google ws2811 led strips to see what this is about... 

//_ on phone

roesl, 2018-01-27:
i dont have to ggogle - when i swith R then R is on  when i Switch G then B and when B then G - so its necessary to change firmware

lobradov, 2018-01-27:
Sorry, I meant legionGer... 

//_ on phone

birca1987, 2018-01-28:
I have WS2811 12v led strip working with latest tasmota on wemos d1. I had to change in xplg_ws2812.ino at line 39 & 312 from NeoRgbFeature to NeoBrgFeature  because the color order on those leds is BRG.
I added this led configuration to homeassistant with below config:
`- platform: mqtt
  name: Office Leds
  state_topic: "stat/OfficeLed/POWER"
  command_topic: "cmnd/OfficeLed/POWER"
  rgb_state_topic: "stat/OfficeLed/RESULT"
  rgb_value_template: "{{ value_json.Color }}"
  rgb_command_topic: "cmnd/OfficeLed/COLOR"
  brightness_state_topic: "stat/OfficeLed/RESULT"
  brightness_command_topic: "cmnd/OfficeLed/DIMMER"
  brightness_scale: 100
  brightness_value_template: "{{ value_json.Dimmer }}"
  qos: 0
  optimistic: false
  retain: true
  effect_state_topic: "stat/OfficeLed/SCHEME"
  effect_command_topic: "cmnd/OfficeLed/SCHEME"  
  effect_value_template: "{{ value_json.scheme }}"  
  effect_list:
      - 0
      - 1
      - 2
      - 3
      - 4
      - 5
      - 6
      - 7
      - 8
      - 9
      - 10
      - 11
      - 12
  payload_on: "ON"
  payload_off: "OFF"`

roesl, 2018-01-28:
I found in user_config.h following line
  #define USE_WS2812_CTYPE     1                 // WS2812 Color type (0 - RGB, 1 - GRB, 2 - RGBW, 3 - GRBW)
Maybe i only have to change "1" to "0" to Switch between RBG and GRB ?
But how can I choose "RBG" ? - there is no CTYPE for RBG
Maybe its possible to add NeoRbgFeature to Neopixelbus library ( NeoColourFeatures.h) to change easily the RBG Definitions for all types of stripes  ?

birca1987, 2018-01-28:
Don't change anything apart in user_config.h, leave it to 1 as the modification I have done on xplg_ws2812.ino will use the #define USE_WS2812_CTYPE 1

ItownTech, 2018-03-17:
I met the same problem.

arendst, 2018-03-17:
Will provide selection for BRG in next pre-release.

Be aware that selections in user_config.h will change from numbers to NEO_RGB, NEO_GRB, NEO_BRG, NEO_RGBW, NEO_GRBW.

skatun, 2018-07-25:
Maybe we can make WS2811 a valid selection type in the config page?
WS2812(RGB800kHz)
WS2811(RGB400kHz)
WS2811(BRG400KHz)
......... and so on, just an idea

Just a final question, if we map a button to one of the GPIO, can this then control the ledstrip(ON,OFF) if the MQTT goes down, i.e manual overide. It would be a nice a feature.

skatun, 2018-07-31:
I just got a few WS2811 led strip of ebay now, and all these are also RBG, so maybe this is the standard way for ws2811? @arendst should we add WS2811 as valid sensor type in the configurator, where the only differnese is what is written in the user config?

daninfuchs, 2018-12-25:
So, trying to pull off a last-second overnight Christmas "decorate the everything with surprise LEDs nobody could know I had until tonight" but apparently the witless folks who made my sets of WS2811 strips used GRB order. Just an FYI that it's out there. Single use-case, I know.

KDMcMullan, 2020-01-28:
I've been working with the 24 LED WS2812-24 ring and it works fine. I just swapped it out for a 50x LED WS2811 set and they appear to have Blue as OK, but Red and Green swapped.

I did a trawl and noticed that the WS2812B protocol specifies the transmission order GRB (MSB first) whereas the WS2811 expects RGB (MSB first). As such, I would expect that a WS2811 receiver receiving signals from a 2812 software driver would have Red and Green swapped, which is what I observed.

I haven't found a software explanation for the OP's problem (G and B swapped).

How do I swap R and G? Must I modify the firmware?

ascillato, 2020-01-29:
> How do I swap R and G? Must I modify the firmware?

No, you need to change in the my_user_config.h file the order of color for WS2812 and recompile. If you need help on that, please, address this to the Tasmota Support Chat. Thanks.

KDMcMullan, 2020-01-29:
TASMOTA support chat isn't going to be able to help me change my_user_config.h without modifying the firmware! I understand what I have to do. Thank you.

ascillato, 2020-01-29:
> TASMOTA support chat isn't going to be able to help me change my_user_config.h without modifying the firmware

Sorry, I don't understand what you meant. There is no need to modify the firmware. Color order is already an option in the config file.

KDMcMullan, 2020-01-29:
Modifying the config file causes a change to the resultant binary. To the software engineer, this constitutes a firmware modification. Its not worth falling out over: I've got my answer and I'm happy. Thanks again.

madtitan77, 2020-02-29:
@KDMcMullan Does it finally work by just
#define USE_WS2812           // Add Shutter support (+6k code)
#define USE_WS2812_HARDWARE  NEO_HW_WS2812     // Hardware type (NEO_HW_WS2812, NEO_HW_WS2812X, NEO_HW_WS2813, NEO_HW_SK6812, NEO_HW_LC8812, NEO_HW_APA106)
#define USE_WS2812_CTYPE     NEO_RGB

in user config?

REALogics, 2020-03-20:
Hi @arendst 

Just curious, couldn't the NeoPixel color sequence (RGB, BGR... etc) instead be configurable in the WiFi interface Generic -> Configure Module -> RGB or BGR... etc... Rather than compiling a fixed sequence in binary for ws2812?  Or it could be done with a Console Command to change the color sequence parameter.  Maybe this would be a lot of trouble coding wise... don't know.

I have designed a few addressable LED WiFi boards for lighting counter tops.  These boards use LED's from China like APA106, WS2811, WS2812d-F8, and others.  These can use different RGB sequences than the standard WS2812.  As a result, the color sequence needs to be compile into the binary for these LEDs.  OverTheAir updating from Tasmota website then overwrites the compile changes.  But, if the changes were done through Console Command or Configuration Module, then update directly from Tasmota would be fine, since config data is not overwritten.  Obviously if the color sequence was done by console command, then only one type of addressable LED could be used for that "Generic" module setting.

s-hadinger, 2020-03-20:
Isn't `SetOption37` solving this now? I don't have those kind of leds so I can't test.

senfkorn, 2020-04-17:
I can confirm, Setoption37 does the job

REALogics, 2020-04-21:
Sorry for late responds.  Before knowing about the SetOption37, I had custom compile Tasmota binary to correct RBG order for my LEDs.  After loading unmodified Tasmota.bin, using "SetOption37 25" on my custom WS2812D-F8 LED Ring does the job.  Amazing and awesome software @arendst, thank you for making this available open source.

xzaz, 2020-04-25:
To be clear: ` SetOption37 7` works for me in combination with WS2811.