arendst/Tasmota #1691: Powering Sonoff Basic with 5v supply

b1g1an, 2018-01-24:
Found this diagram on the Itead website for the original TH showing it being powered by a 5v supply connected directly to two solder points on the circuit board...

https://www.itead.cc/wiki/images/b/b6/Sonoff_th_wifi_switch_wiring.jpg

Given this looks as good as identical to current Basic, those two points are present, anyone got an opinion, or even better experience, on connecting a Basic up like this?

I'd be using it to switch a single 12v line in the same way as shown in that diagram and getting the 5v from a voltage regulator tapped into the 12v.

Some may say the SV is designed for this but I already have a spare Basic and obviously it comes in a box already.

Sky1980chs, 2018-01-24:
I have used a mains/AC connected Sonoff Basic to switch DC through the output relay.
To do this you need to cut pcb traces on both sides of the pcb to fully isolate the relay secondary from mains.  You also need to cut both sides of the pcb to isolate neutral.  Lastly, you also need to add a jumper to the relay secondary.

It is a 5min. / $0 mod., but you need to be very sure you have cut the pcb traces sufficiently as to not feed mains to the relay secondary.

Jason2866, 2018-01-24:
For that usage a sonoff device (in my humble opinion) is overkill.
The advantage off Sonoff is the included switching power supply...
For that usage i would buy a cheap esp wifi relay module.
For example:
https://www.aliexpress.com/item/New-Arrival-1Pcs-Green-ESP8266-10A-220V-Network-Relay-WIFI-Module-Input-DC-7V-30V-Modules/32762870093.html?spm=2114.search0104.3.1.26a8885ebkuG0s&ws_ab_test=searchweb0_0,searchweb201602_4_10320_10321_10065_5920015_10344_10068_10342_10547_10343_10322_10340_10341_10548_10193_10194_10084_10617_10616_10083_10618_10304_10615_10307_10302_10180_10313_10059_10184_10534_100031_10319_10103_441_10624_442_10623_10622_10186_10621_10620_10142,searchweb201603_36,ppcSwitch_5&algo_expid=6a1c4b67-fbd8-4e48-be32-f203f06c472e-0&algo_pvid=6a1c4b67-fbd8-4e48-be32-f203f06c472e&priceBeautifyAB=2

b1g1an, 2018-01-24:
Thanks guys,

@Sky1980chs  that was my original plan but seeing that diagram made me think I could produce a cleaner independent solution.

@Jason2866 overkill possibly but not much cheaper and no case included.

easyyu, 2018-03-13:
@b1g1an 
This link you posted, and the third picture, powering Sonoff basic with 5V istead of 220V is correct?
Did u try?