arendst/Tasmota #1692: Can't compile 5.1.4 without webserver

pascalsaul, 2018-01-24:
Compiling .pioenvs\sonoff\src\sonoff.ino.o
C:/Users/%user%/Downloads/v5.1.4/Sonoff-Tasmota-development/sonoff/xsns_interface.ino: In function 'boolean XsnsCall(byte)':
C:/Users/%user%/Downloads/v5.1.4/Sonoff-Tasmota-development/sonoff/xsns_interface.ino:137:7: error: 'WebServer' was not declared in this scope

WebServer->sendContent(mqtt_data);

Causes by:
//#define USE_WEBSERVER                            // Enable web server and wifi manager (+62k code, +8k mem) - Disable by //

arendst, 2018-01-24:
Forgot it. You solved it. Great. Will upload the fix tomorrow.

pascalsaul, 2018-01-25:
Also with #define USE_DS18x20

Disabled webserver:
stat/th_dbe02/STATUS10 {"StatusSNS":{"Time":"2018-01-25T14:58:59"}}

Enabled webserver:
tele/th_dbe02/SENSOR {"Time":"2018-01-25T15:01:34","DS18B20":{"Temperature":19.3},"TempUnit":"C"}

Similiar bug but quite nasty :) Cold home today ;)

arendst, 2018-01-25:
I don't see the DS18B20 problem. Without webserver compiled in it still reports it's temperature:
```
tele/th1/SENSOR {"Time":"2018-01-25T16:13:25","DS18B20":{"Temperature":41.00},"TempUnit":"C"}
```
measuring the temp of a Sonoff Led.

pascalsaul, 2018-01-25:
All my TH's don't detect the DS18B20's after reloading with a version with the disabled webserver. I'm using the xsns_05_ds18x20 driver.

arendst, 2018-01-25:
Try if this solves your problem:
```
      case FUNC_INIT:
        Ds18x20Init();
        break;
      case FUNC_EVERY_SECOND:
        if ((Settings.tele_period - tele_period) &3 == 3) {
          Ds18x20Convert();  // Start conversion every four seconds, takes up to one second
        }
        break;
//      case FUNC_PREP_BEFORE_TELEPERIOD:
//        Ds18x20Convert();    // Start conversion, takes up to one second
//        break;
      case FUNC_JSON_APPEND:
        Ds18x20Show(1);
        break;
```

pascalsaul, 2018-01-25:
Doesn't solve the problem, still nothing

`stat/th_dbe01/STATUS10 {"StatusSNS":{"Time":"2018-01-25T15:52:56"}}`

arendst, 2018-01-25:
And what happens with the default driver (disable both DS18x20 defines)

arendst, 2018-01-25:
Another option to gain more info is to re-enable the webserver and open the console webpage. If the sensor is there then try a restart using command restart 1 (do not go to the main page and execute restart button as that page initiates the convert again). 

Enable weblog 3 and execute command restart 1. Look for a message like:
```
00:00:00 CFG: Loaded from flash at FB, Count 395
00:00:00 APP: Boot Count 182
00:00:00 Project sonoff Th1 (Topic th1, Fallback DVES_07C8FF, GroupTopic sonoffs) Version 5.11.1e
00:00:00 DSB: Sensors found 1
00:00:00 CFG: Saved to flash at FB, Count 396, Bytes 1648
00:00:00 WIF: Attempting connection...
```
With DSB: Sensors found 1 or 0 in your case.

pascalsaul, 2018-01-25:
With the default driver it does work for one sensor:
`tele/th_dbe01/SENSOR {"Time":"2018-01-25T16:02:08","DS18B20":{"Temperature":28.1},"TempUnit":"C"}`

I also have a Dual where it also works:
tele/dual_sw/SENSOR {"Time":"2018-01-25T16:11:41","DS18B20-1":{"Id":"0317018DDEFF","Temperature":43.7},"DS18B20-2":{"Id":"03170190DCFF","Temperature":46.6},"DS18B20-3":{"Id":"031701ACE7FF","Temperature":39.3},"DS18B20-4":{"Id":"041701BE33FF","Temperature":41.0},"DS18B20-5":{"Id":"0417021727FF","Temperature":39.9},"TempUnit":"C"}

The question is why it doesn't work on the 4x TH's I have with 1 or 2 DS18B20's. I'll enable the webserver with weblog 3.

ascillato, 2018-06-02:
Hi @pascalsaul 

If your issue was solved, please close it. Thanks :+1:

closed 2018-06-13 by ascillato2