arendst/Tasmota #1693: KNX Communication

sisamiwe, 2018-01-24:
Hi,
I'm using Tasmota on my sonoff and it works great. Nevertheless I would like to integrate the sonoff more into my knx installation.

Would it be possible to integrate besides MQTT that communication with KNX.
Here you can find the basic module: https://github.com/envy/esp-knx-ip

What you you think?

davidelang, 2018-01-24:
It's unlikely that Tasmota is going to talk a new protocol, code/ram space is 
tight and we are already fighting the limits, implementing a new protocol would 
be very expensive.

you may want to fork Tasmota and replace either the webserver, or mqtt code with 
something that talks your protocol, then you could make it a compile time option 
for tasmota and benefit from all the driver work for sensorts that is happening 
here.

sisamiwe, 2018-01-24:
Ok I understand.
What would be the best way to implement KNX Communication as an option or to replace MQTT by KNX as a fork?

davidelang, 2018-01-24:
start off by forking and start changing things out to get it working. Once you 
have it working, work to make it an option where you can have mqtt or knx and if 
that's not too painful/ugly, it's possible that arendst will accept it into the 
tasmota codebase (assuming knx is a popular enough protocol to bother with)

ascillato, 2018-01-24:
@sisamiwe I'm interested on implementing KNX. I can help you to develop that.

lobradov, 2018-01-24:
Can you share rationale behind the need for this? 
My google-fu might be rusty, but what’s the benefit of this protocol?

ascillato, 2018-01-24:
The KNX-IP is an open protocol that is for distributed control applications. There is no need for hub. The devices communicate between them. Also several conditions (scenes etc) are configured inside each device. The advantage of this, compared to a local MQTT server for example, is that there is no central device, so if any device stop working, the rest of the installation will continue working.

Take this practical example: You have 2 switches for a stair. You can have one device (i.e. sonoff-tasmota) on one switch and another on the other switch and the stair lamp. They communicate to a MQTT server to report their pushbutton status and the MQTT server re-send the command to the other device. If the MQTT server stop working the Stair light will not work anymore.

KNX-IP uses one predefined broadcast IP for all the devices. Also (depending on the brand) some devices talk each other using a mesh wifi network. So, no router, no server.

There are also KNX-TP (twister pair), KNX-PL (power-line) and KNX-RF

The KNX-TP uses 2 wires for power and communication. Is for very realiable applications.

The others have a protocol that reply commands until it is communicated and understood.

The configuration of KNX devices can be made by ETS software that is just one software for the all the KNX manufacturers. Also there is an official configurator for Raspberry PI.

There are some KNX projects with ESP8266.

So, I think should be nice to implement as an option to use KNX protocol on Tasmota for router or for wifi mesh network.

sisamiwe, 2018-01-25:
@ascillato 
That would be great. How are we communicating for that project? Where are you from?

@lobradov KNX is an industry standard for home automation. It has been called EIB in former times.

ascillato, 2018-01-26:
@sisamiwe, I'm from Argentina. I have forked **esp-knx-ip** to my repositories so we can work from there so as to make an optional library for Tasmota.

johannesbonn, 2018-04-02:
Hi,
I am also interested in a knx connection. Any news on this issue?

Greetings

Johannes

ascillato, 2018-04-02:
hi @johannesbonn , I'm still working on it at https://github.com/ascillato/Sonoff-Tasmota_KNX

hope to have a working version in few days.

After some tests I'm going to make a pull requests to be added to tasmota

ascillato, 2018-04-09:
I have submitted the pull request ##2402 to Add KNX Protocol support to Tasmota.

Surely @arendst will take some time to review it and apply his amazing coding optimizing magic (looking forward to that), so for trying now KNX on Tasmota there is a [Sonoff-Tasmota_KNX](https://github.com/ascillato/Sonoff-Tasmota_KNX) repository that is the actual Sonoff-Tasmota with the KNX Driver.

Enjoy.

ascillato, 2018-05-15:
Hi, This issue is now solved. Please close it. Thanks :+1:

abner0007, 2019-06-23:
@ascillato Could you develop esp 8266  knx  diver？

ascillato, 2019-06-23:
Tasmota already supports KNX. Please, read the wiki on how to use it. This issue is very old. KNX driver was developed and merged a lot of time ago. In the releases page there is also the precompiled Tasmota version with KNX