arendst/Tasmota #1695: Assistant with configuration for buttons and switches on a sonoff basic.

bensuffolk, 2018-01-24:
This is my requirement, but I can't quiet figure out the correct combination of options to get what I want, or it might not be possible but thought I'd check before I dive into the code looking.

A quick bit of background, in my bedroom I currently have a B1 light int he main light, and a standard bulb in the wardrobe. They both come on with the main switch, but the wardrobe has a second switch in series so you can have the main light on and not the wardrobe. 

Since I'm having a B1 my intention is to leave that powered up all the time of course, and just switch the wardrobe light. I'll do a bit of re wiring to sort that out. 

So I'm going to replace the main light switch with a switch panel with 3 retractive switches (operate like push buttons) and a sonoff basic. The relay will control the power to the wardrobe light. I'll connect the 3 retractive switches to GPIO lines. 1 of the retractive switches is actually an up / down switch I plan to use for some dimming of the B1

I initially have tried setting them to switch1 and switch2, switch3 and using switchMode 3. At this point they all control the relay. Toggling it when any button is pressed. Thats the functionality I want for switch1, but not the other 2. 

For switch 2 & 3 I want it to send something over MQTT when I press is, something else when I double click it and a 3rd thing when I hold it. That looks like it should all be possible from the docs if I set it to a button. However nothing seems to happen on a button (even if I set a buttonTopic).

If I set a switchTopic, then all switches send MQTT data, and the first will toggle the relay, but the second & third throws an error message as its also picked up my the device which says there is no realy2/3 (which of course the isn't).

I could just have the switchTopic different and have all the messages sent to HASS and have it send a command back to the relay to turn it on. But I'd still like the basic functionality to work if HASS or the MQTT server is down. So I'd like the switch to toggle the relay straight off.

Is this possible?

Ben

davidelang, 2018-01-24:
you cannot set different switches with different switchtopic settings, you will 
have to have them all send mqtt messages and your home automation software send 
back the command to switch the local relay.

bensuffolk, 2018-01-24:
What about the push buttons and buttonTopic, is that not different to the switches?

Ben

bensuffolk, 2018-01-24:
What about buttons are they different to switches? I couldn't get the inputs set as buttons to do anything though even with the button topic set.

Ben

davidelang, 2018-01-25:
buttons are separate than switches, but buttons can only be momentary 'toggle' 
functionality.

bensuffolk, 2018-01-25:
That's fine, I'm using retractive switches (spring loaded - could also be called momentary or push buttons). But how do I get them to work, setting up button2 with a ButtonTopic gave no results.

bensuffolk, 2018-01-25:
I'm thinking thats its because on sonof.ino ButtonHandler only loops over devices_present as opposed to all possible buttons

```
  uint8_t maxdev = (devices_present > MAX_KEYS) ? MAX_KEYS : devices_present;
  for (byte i = 0; i < maxdev; i++) {
```

I notice SwitchHandler loops over all possible switches though.

```
  for (byte i = 0; i < MAX_SWITCHES; i++) {
```

Is this intentional or a bug in ButtonHandler?

davidelang, 2018-01-26:
you set buttontopic for the entire device, not for one button

so it would be buttontopic foo not buttontopic2 fo

bensuffolk, 2018-01-26:
@davidelang yes I saw it was an all or nothing thing, but it looks like I can only set up a button if I have a relay, unlike a switch that can be setup without a corresponding relay (See above for the code snippet).

But I think, please correct me if I'm wrong, that I can only have Hold and double click on buttons not switches.

Ben

bensuffolk, 2018-01-26:
Ok,

So I have done a quick mod to change the ButtonHandler to loop through MAX_KEYS rather than devices_present so I can see what happens.

It seems that no matter what buttons are configured or pressed its always based on POWER not the button number. If you configure relay2 on the output then you get POWER1 and POWER2 send over the MQTT ButtonTopic. 

But doing this means I loose a GPIO line as it has to be assigned to a relay. Since the BASIC has 4 configurable GPIO lines, it means I could only have 2 extra buttons (as I'd need 2 extra relays). I think thats not a good idea.

In ButtonTopic mode when its sending commands, I see no reason why it can't just send a command based on the button you press (after all I believe the main idea of using a Topic is to send commands to a different device, not back to it self).

So I think that if you have a ButtonTopic configured the correct behaviour would be to send an MQTT message based on the button number pressed regardless of the number or state of local relays.

@arendst  does that make sense or have I completely mis-understood the purpose of Buttons and Topics?

davidelang, 2018-01-28:
hold and double click are only features on button0 as far as I know

danstoian, 2018-03-29:
@bensuffolk I did had the same problem with multiple switches and I solved it using a virtual relay connected to GPIO4. 

In my current setup I do have a sonoff basic mounted in-wall on the hallway. The main switch (Switch1) controls the light on the hallway. I also needed a secondary switch (momentary) connected as Switch2 to turn off all lights when I leave. 

It´s important to mention that the Switch2 (momentary) does change the state of the Relay2 however the Relay1 state is not changed. Both switches are sending messages on MQTT. State is also reported correctly.

Here is how I solved it...

1. Module configuration (01 Sonoff Basic)
GPIO1 Serial Out = 01 DHT11
GPIO3 Serial In = 09 Switch 1 (Main switch that controls the relay)
GPIO4 = 18 Relay2 (GPIO4 is not connected anywhere)
GPIO14 = 10 Switch 2 (TTP223 capacitive touch sensor)

2. SwitchMode/SwitchTopic configuration
Switch1 - Used default topic and default mode. SwitchMode1=0 (TOGGLE)
Switch2 - Used default topic and FOLLOW mode. SwitchMode2=1 (FOLLOW)

Now I´m able to switch all the lights off by touching the capacitive sensor for 5 seconds (automated in Hass) and also use the existing switch to turn the hallway light on/off. 

[image]

[image]

Tousakusha, 2018-05-31:
This is exactly what i was looking for. But i have a problem.
As soon as i set any GPIO to 09 Switch 1 - my dht11 will stop sending mqtt messages to my iobroker.
What am i doing wrong?!

danstoian, 2018-05-31:
@Tousakusha I don't know exactly why the sensor stops working. I do have the DHT11 connected to GPIO1 (TX) using a pull up resistor (4K7 - 10K) between the VCC and DATA pins. I have no problems it (no spikes or mqtt loss). On another module I do have a AM2302 (DHT22) connected without any resistor on the same port. My wires are normal breadboard jumper wires about 10cm long.

I've set the module to sleep for 100ms, SensorRetain to ON and TelePeriod to 60s but I'm not sure if there is any relevance:
 - http://<module_ip>/cm?cmnd=Sleep%20100
 - http://<module_ip>/cm?cmnd=SensorRetain%201
 - http://<module_ip>/cm?cmnd=TelePeriod%2060

Tousakusha, 2018-05-31:
@Danstoian Thanks for your reply! My German Firmware seems to have some weird settings.

Stating my Problem in english made me think of changing to the original Tasmota.

It works now. Thank you! :)

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2