arendst/Tasmota #1696: Issue with freshly flashed Sonoff

legionGer, 2018-01-24:
Hey guys,

I recently flashed my first Sonoff. No issues. Accessing the WIFI-configuration page was easy. Sonoff connected to WIFI. All nice. Now the trouble starts. Despite immediately showing up on my router, the device can most of the time not be accessed via WEB. It only shows "Webpage cannot be displayed. Response time too long". I checked it via CMD and ping, 50-75% of packages are dropped. After seperating from a power for a bit now the switch itself goes crazy. It won't react to any button presses, instead it always on but will randomly toggle the relay off and quickly on again. 

Can someone help me out?

davidelang, 2018-01-24:
try holding the button down for 4 seconds to reset it's config to what's in the 
compiled default.

legionGer, 2018-01-24:
I tried that already. It is not even responding to that. It appears to just ignore any button presses. It just keeps randomly switching the relay and led briefly.

ascillato, 2018-01-24:
can you see the SERIAL LOG? sometimes it hungs when trying to connect to an inexistent MQTT server. try to turn MQTT off on the code before flashing

legionGer, 2018-01-25:
I sadly don’t have access to any logs at the moment. Web is not accessible and I don’t have the USB TTL Adapter as mine is still being shipped and a friend flashed it for me. You reckon I should just flash it again with MQTT disabled? 

Could the issue be with the router or the hardware of the sonoff?

ascillato, 2018-01-25:
You can try. After flashing, if it works, you can turn it on again by web.

legionGer, 2018-01-25:
Could this also be a memory issue? As far as I know be pretty much flashed the Default Version on a Basic Switch. I found some older threads with similar issues where flashing the minimal version helped. 

So maybe also try removing some stuff like TLS or homeautomatic Support from the firmware?

svenwhd, 2018-01-25:
I've what I think is the same problem, but I managed to identify it as an issue with the web interface, which appears to crash if I use the main menu (no issue in at least some sub menus like the configuration). This leads to a reboot of the Sonoff, sometimes in rapid sucession. If I don't touch the web interface at all, everything works stable and fast, including switching by MQTT.

legionGer, 2018-01-26:
So I reflashed with MQTT disabled. Issue still persists. Before connecting the Sonoff to my WLAN it responds fine. As soon as it connects, it wont react. Pretty much no connection on the Web Interface, no reaction to button presses at all. Anyone more ideas?

curzon01, 2018-01-27:
Which version do you are using?
What is the output of "Status 2"

I can confirm the behaviour you originaly described (web not responding after a while, mostly works directly after a restart, later not) when I compile 5.11.xy versions using core lib 2.4.0.

legionGer, 2018-01-27:
For me it was some issue on the 5.11.1d firmware. I flashed the 5.11.1 over OTA and now it works just fine.

davidelang, 2018-01-28:
what shows up in the logs? set the log level to 'more detailed'

curzon01, 2018-01-28:
@legionGer:
What shows "Status 2" on Console with 5.11.1d where it didn't worked. Is this a firmware with core 2.4.0? If you have compiled it by yourself be aware which library you are using - see also #1706

legionGer, 2018-01-28:
@davidelang I wasnt able to access the logs while connected to WLAN. Sonoff did not respond. 
@curzon01 I don't know sorry. A friend flashed it for me.

closed 2018-01-28 by legionGer