arendst/Tasmota #1698: Support for Wemos D1 mini v3

mpuff, 2018-01-24:
Hello,

therre is a new wemos d1 mini v3.0.0.
The frimware doesnt work with that, it is also esp8266 based.
Maybe you can add to support this.

Frogmore42, 2018-01-25:
What is not working?
Looking at the schematic for the v3, it is basically the same and should work fine.

mpuff, 2018-02-05:
i have now seen in a datasheet that hte v3 is  ESP8266EX based!

Igor122017, 2018-02-05:
hello? i have  a new wemos d1 mini v3, load is ok but do not working...

henkez73, 2018-02-05:
@Igor122017 That looks lime a lolin to me. Try flashing it as a nodemcu instead of wemos di mini.

Frogmore42, 2018-02-05:
The esp8266EX is the current full name for esp8266 (from what I have read) and has been all that is available, ie is fully compatible with Tasmota.

If you are having an issue, please provide more details about exactly what you did and what is wrong.  Saying it is not working is not likey to lead to a solution.  Providing the console/serial output might.

Igor122017, 2018-02-05:
@ Igor122017,  Thanks you very much... all working... i am uncomment out the line in platformio.ini that says:
upload_resetmethod = nodemcu
tell me please, where i can find name pin?

mpuff, 2018-02-05:
today i got another v3, this one is working.
thank you for your help

closed 2018-02-05 by mpuff