arendst/Tasmota #1700: Belkin WeMo Emulation don't inform change status

ascillato, 2018-01-25:
When I activate **Belkin WeMo Emulation** on Sonoff Tasmota v5.11.1e it get discover very fast on _Home Assistant._

From HA I can turn it ON and OFF, but if I press the button on Sonoff or change the state from MQTT for example, the change is not shown on the card of Belkin WeMo on HA.

On Sonoff's console shows the reception of the command from Belkin WeMo of HA but says nothing of sending any response to HA. Console is on DEBUG 4.

[image]

On the screenshot can be seen that I turn Sonoff On by Wemo on HA and then I turn it off by button on Sonoff. MQTT is also configured and it receive the tele as expected.

[image]

How I can make Sonoff inform the Status over Belkin WeMo Emulation?

ascillato, 2018-01-28:
On Home Assistant Log, nothing is received when pushing the button on Sonoff.

Home assistant just receives the status when you toggle on Home Assistant.

ascillato, 2018-01-28:
I tried other firmwares for emulating a WeMo Device like [espurna ](https://github.com/xoseperez/espurna) and the behavior was the same on Home Assistant and on Wemo android App.

So, @arendst is this behaviour (not inform the change of status) a bug on the firmware or is a normal behaviour on Wemo Devices?

handsomejackuk, 2018-02-08:
can anyone confirm if wemo emulation works i have entered my sonoff mac address and my router name in connect wemo to google assistant, on my android phone but cant get past the submit button, please advise, tasmota   5.11.1   2018-01-07T21:51:09

ascillato, 2018-03-06:
Hi @handsomejackuk, the WeMo Emulation works. The only issue is that if you change the sonoff status locally (on the sonoff button) it don't inform this change to home assistant. If you use Hue Bridge Emulation it does inform the change.

@arendst, this behaviour (not inform the change of status) is a bug on the firmware or is a normal behaviour on Wemo Devices?

ascillato, 2018-04-19:
Could not find a solution. Closing...

closed 2018-04-19 by ascillato