arendst/Tasmota #1703: Error compiling for board Generic ESP8266 Module.

elder1104, 2018-01-25:
I am new to flashing Sonoff boards, I am getting the following error. Any help is greatly appreciated.

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 1M (64K SPIFFS), ck, Disabled, None"

In file included from C:\Users\Dad\Downloads\Sonoff-Tasmota-5.11.1\Sonoff-Tasmota-5.11.1\sonofftest1\sonofftest1.ino:35:0:

sketch\i18n.h:326:30: fatal error: language/en-GB.h: No such file or directory

   #include "language/en-GB.h"

                              ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

gajotnt, 2018-01-25:
Read the wiki 1st....

NeildD1968, 2018-01-28:
I have the exact same problem.

Maybe I've misinterpreted the tone, but "Read the wiki 1st...." seems smug and unhelpful to me.

JuliusLedoux, 2018-02-07:
I am having the same problem. I am trying to flash a Sonoff SV. No luck here, very frustrating! Ans yes I read the wiki and also I have done 4 different sonoffs (basic) with a previous version of the firmware with no issues.

morgapa, 2018-03-15:
same problem 

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 1M (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build options changed, rebuilding all
In file included from sketch\config/all.h:28:0,

                 from C:\Users\morgapa\Desktop\arduino-1.8.5\espurna-dev\code\espurna\espurna.ino:22:

sketch\config/hardware.h:1655:6: error: #error "UNSUPPORTED HARDWARE!!"

     #error "UNSUPPORTED HARDWARE!!"

      ^

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

lobocobra, 2018-04-04:
You probably want to read following page....
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

And in detail following sentence.....

> For all versions
> Copy all files from the Tasmota release Source code folder lib into your sketchbook\libraries folder.
> Copy the Tasmota release Source code folder sonoff to your sketchbook.

Rishi159, 2019-03-30:
C:\\Users\\abcd\\AppData\\Local\\Temp\\arduino_build_567397\\sketch\\Blink.ino.cpp" -o nul
xtensa-lx106-elf-g++: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board Generic ESP8266 Module.
I've tried almost everything but this error won't go away. Shall I just buy a NodeMCU