arendst/Tasmota #1704: Error Upload Firmware

rodolfoPlanetfone, 2018-01-25:
Hello friends. Sorry for bad English.

I was performing the process of changing the firmware of a Sonoff Basic that I purchased.

I followed the recommended tutorial on the wiki project -> https://www.youtube.com/watch?v=n4MDRm2yAJg, but when I finished compiling the following error occurred while uploading.

`warning: espcomm_sync failed`
`error: espcomm_open failed`
`error: espcomm_upload_mem failed`
`*** [upload] Error 255`

I looked for solutions at Google, where some users asked for change at TX / RX and RX / TX reverse for RX / TX and TX / TX, but the problem persisted.

After several attempts the error persisted. Now, I'm trying to put Sonoff on the power and your Led is not going up and I think it's dead.

Here are some images of the process and error. Any help would be very welcome. Thanks.

https://ibb.co/dxOKuG
https://ibb.co/nqZ3Mw
https://ibb.co/dN5oob
https://ibb.co/gNsq1w

gajotnt, 2018-01-25:
Are you setting the sonoff to programming mode?

rodolfoPlanetfone, 2018-01-25:
Yes, I'm setting the Sonoff for programming mode

rodolfoPlanetfone, 2018-01-26:
Update

i try configure Esptool for Tasmota upload

During the upload process I get the following error.
`a fatal error occurred failed to write compressed data to flash after seq 0 (result was c100)`

I think the problem may be my USB connection. I am using a data cable of the motorola g.

My FTDI converter
https://http2.mlstatic.com/conversor-usb-serial-rs232-ftdi-ft232rl-ttl-5v-3v3-arduino-D_NQ_NP_161825-MLB25490970661_042017-O.webp

Problems that are occurring
https://ibb.co/csveeG
https://ibb.co/mVqZDb
https://ibb.co/mPfCzG

rodolfoPlanetfone, 2018-01-26:
I found the problem

It really was on the USB connection.

I released the serial port from my raspberry pi 3 and did it for him without the need for an FTDI converter.

closed 2018-06-10 by stale[bot]