arendst/Tasmota #1706: Console in version 5.11.1e does not work

chkbln, 2018-01-25:
Hello,

with version 5.11.1e the console does not work anymore.
No messages are displayed, for example, from the MQTT.
Input commands do not produce output in the console.
Version 5.11.1d is fine

greetings

arendst, 2018-01-25:
It now uses chunks. 

Pls provide output of command status 0. What version of esp/arduino do you use? What browser?

auno88, 2018-01-25:
It happens the same to me. The console does not work and it works extremely slow. I have returned to version 5.10 and everything works correctly. I use basic sonoff and it happens to me with all browsers.

chkbln, 2018-01-25:
IPhone7 Safari does not work

Win 10 IE works but delayed, reports Komamndos delayed 3-5 sec. Back

Win 10 Firefox 58 does not work

Win 10 Edge works after about 1 minutes (telemetry period = 60) the messages come, commands are not returned. Back to the main menu, then back to the console. Console empty. With the telemetry the entries come again

Androit Chrome like Win 10 Edge

Hardware Wemos Mini Clone

arendst, 2018-01-26:
What version of esp/arduino do you use? 2.3.0 or 2.4.0?

arendst, 2018-01-26:
Just released version 5.11.1f uses the "old" full webpage again and should solve the issue noted here.

I did notice that Microsoft Edge needs at least 9 characters on it's console input field before taking action after hitting return. This needs more investigation but workaround is entering extra spaces or hit the return key twice ;-).

curzon01, 2018-01-26:
Console works on 5.11.1f with browser where the webinterface works (here for example Chromium  63.0.3239.132 (official build) built on Ubuntu)

But what's the reason why webinterface does not work in general on branch "development" using Core 2.4.0/SDK 2.1.0?

- On my smartphone there are often no response on http port 80 and if responding, the page is completely empty.
- On Desktop browser sometimes there is garbadge on screen because the responding data does not contain the complete html text.

arendst, 2018-01-26:
Lack of ram result in incomplete pages and empty screens. This is more obviouse now as the mqtt buffer has been increased using more ram to provide hass mqtt discovery. Using 2.4.0 makes things worse as it uses more ram and is experimenting with LwIP versions.

Try 2.3.0 for a more robust experience.

emontnemery, 2018-01-26:
@arendst, how much extra ram is used by 2.4.0?

chkbln, 2018-01-26:
Hello
with version 5.11.1f the console works again.
I use arduino with 2.3.0

Thanks and regards

arendst, 2018-01-27:
@emontnemery ram usage is a bit dynamic. What others have noted too is that after a minute or 10 ram usage settles some 4k less than with 2.3.0. This happens because of logging which assigns up to 20 ram slots per log line of max 400k. Below 14k free heap it becomes a problem to display the Configure Module webpage for a Generic (former Wemos d1 mini) module displaying all GPIOs. Although the webpage is about 6k3 is seems to me the webserver is allocating memory again to serve the page.

emontnemery, 2018-01-28:
@arendst I pushed a suggestion for non-dynamic log buffer in #1730 to avoid system memory use varies due to log content

arendst, 2018-02-03:
Changes implemented in 5.11.1g

closed 2018-02-03 by arendst