arendst/Tasmota #1707: sonf of touch  connect bug?

vMeph, 2018-01-26:
ok i got the sonoff touch working with homeassistant and retain states with the PowerRetain command
so if i restart homeassistant it will get the states of lights from before
now the problems is if you reboot host hassio(HA), then the sonoff cant reconect with mqtt HA, with out restart himself.
if you restart sonoff then it will get the states of lights from before..
the question is why cant sonoff reconect with the host with out restart??
is this a bug?

ascillato, 2018-01-27:
Hi,

Is not a bug. It is a config problem on Home Assistant (Hass.io). I had the same.

I put the following configuration on Home Assistant and worked no matter which device you restart.

on _configuration.yaml:_

```
# switch
light:
    - platform: mqtt
      name: "NodeMCU Test"
      state_topic: "stat/sonoff/POWER"
      command_topic: "cmnd/sonoff/POWER"
      availability_topic: "tele/sonoff/LWT"
      qos: 1
      payload_on: "ON"
      payload_off: "OFF"
      payload_available: "Online"
      payload_not_available: "Offline"
      retain: false
```

vMeph, 2018-01-27:
@ascillato it seems that with that if i do a hassio host reboot and ligths was on when it comes back light will be switch off, isnt retaining flags

another problem is why doesnt sonoff toutch automatic log into moskito fast?
it takes forever to log in, e have a ESP project that automatic logs in, so whats the problem here

i had this code working:
- platform: mqtt
    name: "testl"
    state_topic: "stat/sonoff/POWER2"
    command_topic: "cmnd/sonoff/POWER2"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    optimistic: false
    retain: true

now i went to try you code and is back to the same problem :(
when logs in the lights that was on butons goes off but lights are on

my question is why works perfect with ESP with never problems just by doing:
  digitalWrite(test1,LOW);  
  client.publish(MQTT_STATE_TOPIC_test.c_str(),"OFF",true);

and why doesnt work with tasmona
makes me wonder if is a tasmona problem or a homeassstant problem
with regular ESP projects there isnt problems with HA

ascillato, 2018-02-17:
Hi

Could you resolve the issue? 

With the configuration shown, the one that retains the state, in my case, is tasmota. Thats why if i restart my sonoff will be on the previous state. And if i restart home assistant, it will read the tasmota state without changing the state.

If this is not working may be a config in your mqtt broker. I'm using the embedded on home assistant and it is very fast. Also tasmota logs very fast on it. 

If I unplug and plug again the sonoff it takes like 1 to 2 seconds to connect to wifi and show it status on home assistant.