arendst/Tasmota #1708: Hostname not reporting to DHCP server correctly

Anx2k, 2018-01-26:
I'm running the latest version (5.11.1) and when I change the hostname to something like "Outlet40", on my dhcp server (pfSense 2.4.2-RELEASE-p1) just shows "4".  Many times it just shows a blank entry and doesn't show anything related to the hostname specified.  I tried resetting to factory defaults and the problem still persists.

ionciubotaru, 2018-01-27:
maybe it's a pfSense bug, I'm using isc-dhcp-4.3.5 on latest raspbian with Tasmota 5.11.1d ant it reports fine:
```
lease 192.168.1.29 {
  starts 6 2018/01/27 12:58:14;
  ends 6 2018/01/27 13:08:14;
  cltt 6 2018/01/27 12:58:14;
  binding state active;
  next binding state free;
  rewind binding state free;
  hardware ethernet 60:01:94:85:36:74;
  client-hostname "mt_copii-5748";
}
```

Anx2k, 2018-02-10:
For what it's worth, with the new version (5.12.0) I'm not seeing this at all - everything works as expected.

closed 2018-02-10 by Anx2k