arendst/Tasmota #1709: What is the cable called to connect it to the PC?

williamrussellajb, 2018-01-26:
Hello, 

I want to try this project as a hobby. I watched a course about arduino boards on lynda.com (So I'm super new). 

What is type of cable that is being used to flash the sonoff? The cable connecting the pins on the board to the PC.

Thanks!

williamrussellajb, 2018-01-26:
I am about to buy this, but I want to be 100% sure.

https://www.adafruit.com/product/954

ionciubotaru, 2018-01-26:
It's not good because it's a 5V FTDI and sonoff needs 3V3
https://www.banggood.com/FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-For-Arduino-p-917226.html?rmmds=search&cur_warehouse=UK This has a jumper to choose between 5V and 3V3

williamrussellajb, 2018-01-26:
Hello! Thank you very much, I bought this and the 120 pin cable connector.

williamrussellajb, 2018-01-26:
@ionciubotaru looking at this image: 
https://drive.google.com/file/d/0Bw1XOPtMPlsNQmxkTjNfa0xreVk/view

Do I need to have an additional power supply to the bottom two red circles?

ionciubotaru, 2018-01-26:
no, you don't need to supply power to the bottom connectors, supplying power to ftdi power pins it's enough
Connect 3V3, TX,RX,GND to ftdi, ignore GPIO14

closed 2018-06-10 by stale[bot]