arendst/Tasmota #1710: Sonoff basic firmware upgrade webinterface

marcgrc, 2018-01-26:
hello everyone I have a problem.
I can not update the firmware with a web interface.
the line that gives is this
http://domus1:80/api/arduino/sonoff.ino.bin
 but when you restart the version remains the same.
thank you so much

ionciubotaru, 2018-01-26:
Try to download sonoff.ino.bin from your computer using chrome or other browser to check if the address is correct.

marcgrc, 2018-01-26:
from where I unload sonoff.ini.bin ??

ionciubotaru, 2018-01-26:
from http://domus1:80/api/arduino/sonoff.ino.bin 
I'm not sure this address is correct - if you are not familiar with web server try to upload the bin file via web interface manually

marcgrc, 2018-01-26:
Ok thank
http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin
It’s ok
I have another question 
Before to flash sonoff basic 

just bought you have to do something ??

SodoffBaldrick, 2018-01-26:
If it's OTA then no. Just go for it. 
Perhaps back up your settings if you're at all bothered by that, but by default it won't overwrite any settings.

marcgrc, 2018-01-26:
Sorry 
OTA ????

closed 2018-01-27 by marcgrc