arendst/Tasmota #1711: Remote access. Port forwarding to sonoff. No access.

storm66-ag, 2018-01-26:
Hello. I use sonoff tasmota FW. Thank you for it.  Everything is working, but web access works in local network only. If set port forwarding from Internet, using wifi router, that web page is not downloads, but appear login:password window and it's all. No web panel, not menu. But in local lan its available. 
I believe it may latency ????

mzhur, 2018-01-26:
Are you use the DHCP configuration on Sonoff?
I'm not sure, but it should be problem in router firewall probably. 
And one more think, some internet providers close the port 80, and some times ports below 1024 for incoming connection.
Would you try to move the tasmota web server port up to 10 000 in the user_config.h.

storm66-ag, 2018-01-26:
Yes, DHCP configuration used. Is  port 80 only using  in tasmota SW for web access, or same else ports? 
I see the login and password window, after insert I see white  screen. If so the firewall is working correctly and provider is not blocking port 80. May be this from hight latency? I used LTE network with: 
round-trip min/avg/max/stddev = 25.645/180.648/571.087/139.173 ms

mzhur, 2018-01-26:
Ok, first time, i'm not clear understand the situation. I don't know about

mzhur, 2018-01-26:
the latencity influence to the tcp/ip engine in esp8266

storm66-ag, 2018-01-26:
I have WiFi router with Build in lte modem. Sonoff connected to WiFi network, and in local network everything working good. But I need accces to sonoff module from internet. I set port forwarding on router. The router has public IP address. And doesn’t show web page from internet.

mzhur, 2018-01-26:
I try to test it with my sonoff during the weekend...

storm66-ag, 2018-01-27:
Ok, try please.  And use last software please.

storm66-ag, 2018-01-27:
I try with other provider and other router, its everything ok.

storm66-ag, 2018-01-28:
its working on sonoff TH. But for sonoff basic is not working......

storm66-ag, 2018-01-28:
[image]

storm66-ag, 2018-01-28:
This is that I get.

mzhur, 2018-01-29:
I try with my modem and Sonoff Basic. In my case it work, but i have fasters pings...

storm66-ag, 2018-01-31:
With fast ping it’s working very well, but if round trip delay is more than 150-200 ms - it’s not working...

sonoffheld, 2018-02-02:
i have the same Problems with openvpn thats not worked .
with an ssh connection over a tunnel and proxy everything works fine.

closed 2018-06-17 by stale[bot]