arendst/Tasmota #1712: Sonoff *** [upload] Error 255 - any fix yet?

Kepete, 2018-01-26:
I have connected my Sonoff to FTDI 3.3v and try to upload firmware using Platforio. I hold down the button on the Sonoff when connecting for 5sec. Also tried to hold it down until it uploads everyting.
No LED light signals from Sonoff. I have tried swapping the TX and RX cables.

I get the error *** [upload] Error 255. 

I have read the whole internet through and no fix found. What do to?

SodoffBaldrick, 2018-01-26:
I've uploaded using PlatformIO with no problems so don't believe there are any inherrant errors that suggests there is an ongoing issue (as the issut tile implies by using the word "yet" - have you got a link to details of an issue affecting more than just yourself?)

Basic questions but are worth checking:
 - Do you have RX to TX and TX to RX or RX to RX and TX to TX?
 - Have you tried flashing a stock firmware outside of PlatformIO?
 - You only need to close GPIO0 for a second while powering on the ESP8266, are you doing this?

Kepete, 2018-01-26:
Thank you for your comment @SodoffBaldrick ! 
Yes I have seen many others having the similar problem, some links: 
https://github.com/arendst/Sonoff-Tasmota/issues/463
https://github.com/arendst/Sonoff-Tasmota/issues/840
https://github.com/arendst/Sonoff-Tasmota/issues/276

Answering your questions: 
- Yes I have tried both ways, still error persist. 
- No I haven't. Where to start? 
- I have tried closing GPIO0 for a second, for 5 seconds, for the whole time etc

SodoffBaldrick, 2018-01-26:
Sorry, I wasn't aware there was a larger issue, my mistake.

For point 2, try the following

- Download esptool.exe (sorry, I don't have a link, I forget where I got it from originally)
- Download sonoff.bin https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.11.1/sonoff.bin
- Run the command "esptool.exe -vv -cb 115200 -cp COM5 -ca 0x00000 -bz 1M -bm dout -cf sonoff.bin" 
       - of course substituting the path of esp tool, sonoff.bin and also using the correct COM port

rodolfoPlanetfone, 2018-01-26:
Hello Friends,

I'm also having the same problem as @Kepete.

I tried adding a firmware via esptool but an error occurred while uploading

Follows link with some images of the process
#1704

Kepete, 2018-01-26:
@SodoffBaldrick  So I'm using Mac OS and I tried the same steps following mac os tutorials :) 
It went well I think

[image]

After this the green LED is now blinking if I connect again it to computer. How to other verify that it is working correctly? 

What should the next steps be?

SodoffBaldrick, 2018-01-26:
Excellent :)

Do you see a sonoff-#### wifi point on your computer? if so then connect to it and enter your SSID and password and then you should be set :)

Kepete, 2018-01-30:
@SodoffBaldrick  

I was few days away and couldn't test it more. Today got back and now going to play with it again. 
Right now I have it connected with USB to my computer, but no wifi network is showing up. Do i have to connect it to wall socket for the wifi to appear?

closed 2018-01-30 by Kepete

johannarispe, 2018-05-11:
I had a similar issue as above with platformio. I tried it in a macOs High Sierra with thunderbolt3, and it didn't connect the serial communication. Then tried with the esptool script, and also didn't work with that mac. 
It worked smoothly fine using esptool on a windows machine. 
After flashing it, just need to press 4 times to create a sonoff AP and update the network settings.

sweebel, 2018-07-31:
For future people with this problem
Download drivers for mac from http://www.ftdichip.com/Drivers/VCP.htm

solution found at https://github.com/arendst/Sonoff-Tasmota/issues/957

four2six, 2018-12-11:
just for future reference, on a mac with osx if you get the issue COM5 not found, change upload_port = COM5 in platform.io to upload_port = /dev/cu.usbserial-AH00TU08 or whatever you actual port is you find in the serial monitor.