arendst/Tasmota #1713: alexa read temp from WeMos D1

cobra85ec, 2018-01-26:
can alexa read temp from WeMos D1 mini with DHT11

SodoffBaldrick, 2018-01-26:
Not directly I believe. You'd want to have an MQTT broker installed somewhere which is connected to Amazon.
So the WeMos reports the status via MQTT to a server. Alexa is configured to connect to the server and read the variable for that particular sensor.

Have a look into Home Assistant.

cobra85ec, 2018-01-27:
i have MQTT on a pi3 with node-red also...

Jason2866, 2018-01-29:
Install OpenHab on the Pi. With the HueEmulation (you have to install via Paper UI) is this possible.
You have to "Tag" this item... Alexa will discover it...
I use this in my Homeautomation with a DHT22 (more accurate).

closed 2018-06-13 by stale[bot]