arendst/Tasmota #1714: Customizing GPIOs

MacPit69, 2018-01-26:
Hello.

I flashed the tasmota firmware to an ESP8266 based relays from lctech-inc.com
It seems they used another GPIO than SONOFF to switch the Relais.
Is it possible telling the firmware the right GPIO for switching the relais?

thx
MacPit69

SodoffBaldrick, 2018-01-26:
I'm not sure if I completely understand the question, but I'll have a try.

On the web interface click Configure
Click Configure Module

Which version of firmware did you use?
If 5.11.1c or newer then under Module type select Generic
If 5.11.1b or older select Wemos D1 mini.

Click save.

Once its rebooted you'll now see a list of all the GPIOs and you can select the appropriate one and assign it to a relay.

Jason2866, 2018-01-26:
Hello,
lctech-inc.com wifi relay wotks not via gpio. It switches via commands on the serial line.
There is a pull request in which this relay is supported.

KPK75, 2018-01-26:
I did PR for 1 relay version. It won't work without changes in the code. What version do you have?

Jason2866, 2018-01-27:
Hi KPK75,
this one.
https://www.aliexpress.com/item/ESP8266-5V-WiFi-relay-module-Things-smart-home-remote-control-switch-phone-APP/32734997467.html?spm=2114.search0104.3.75.31cf7f44dPzlyy&ws_ab_test=searchweb0_0,searchweb201602_4_10320_10321_10065_10344_10068_10342_10547_10343_10322_10340_10548_10341_10193_10194_10084_10083_10618_10304_10615_10307_10302_10180_10313_10059_10184_10534_100031_10319_10103_441_10624_442_10623_10622_10186_10621_10620_10142,searchweb201603_28,ppcSwitch_5&algo_expid=cf6fc710-d9a9-4484-b117-72d0be1507a1-11&algo_pvid=cf6fc710-d9a9-4484-b117-72d0be1507a1&priceBeautifyAB=2
I unsoldered the Uart chip and connected direct one GPIO chip via the onboard Transistor to the relay.
Added some pullup resistors too. Now it works reliable.
So it is fully compatible to the Tasmota software, without any change in the software...
Just now ordered this one (i need some for a project, so better no or minor changes on the board and the have to be cheap...)
https://www.aliexpress.com/item/ESP8266-5V-WiFi-Relay-Module-Things-smart-Home-Remote-Control-Switch-Phone-APP-ESP-01-Wireless/32842671078.html?spm=a2g0s.9042311.0.0.XIj6Ab

KPK75, 2018-01-27:
The second one works without any problems. AFAIR relay is connect to GPIO0 and blue Led on ESP-01s to GPIO2. There is another blue led on pcb connected directly to relay.

MacPit69, 2018-01-29:
Hi.
This is my one: http://www.chinalctech.com/index.php?_m=mod_product&_a=view&p_id=1204

KPK75 do you have pictures from your Modifications?

thx

KPK75, 2018-01-29:
Mine looks the same. And I only hacked the code.

MacPit69, 2018-01-29:
can I get this hack

KPK75, 2018-01-29:
It's there  #1282. I made PR but it didn't find a way to dev version yet so you have to make changes to sonoff_template.h and sonoff.ino yourself. All changes are on the git.

ddamron, 2018-01-29:
I have tested this and it doesn't work in the current form.  
Everything looks good, but there is one missing item..
in sonoff_template.h, we need to also add an item in kNiceList[MAXMODULE]
add in LCTECH, in the same position as you did in the previous.  Be aware that WHERE you add it matters..

Once added, and recompiled, I now have a whole bunch of LCTECH relays working.

KPK75, 2018-01-29:
Yes. My fault. Thanks. Do you have 1relay version ? Or 2 or 4 ?

mischkasmith, 2018-02-25:
Hello.
I have the 2 channel relay. Can you please add this?
Your code works with the relay 1, by changing the baudrate to 115200.

This model use a STM8S003F3P6
Relay 2 needs  A0 02 01 A3 for ON and A0 02 00 A2 for OFF.

THX.

KPK75, 2018-02-27:
What is the baudrate of 2 channel version ? 9600 like 1 channel ?

mischkasmith, 2018-02-27:
The baudrate of 2 channel version is 115200.

KPK75, 2018-02-28:
Try this repository : [https://github.com/KPK75/Sonoff-Tasmota](https://github.com/KPK75/Sonoff-Tasmota) and test. I everything works I'll make PR to main repository.

salopette, 2018-03-03:
Hello, thanks for the build.
I have a module with 5V, it worked there.
But it does not work with a 12V module. Do you have to pay attention to something?

Thank you

KPK75, 2018-03-03:
Maybe different baudrate? Try change 9600 to 115200.

mischkasmith, 2018-03-04:
Hello, THX for adding the 2 channel version.
But it only switch the relay 2.
I have add a 10 ms delay. Now it works fine.

`      Serial.write(0xA0);
      Serial.write(i+1);
      Serial.write(state);
      Serial.write(0xA0+i+1+state);
      Serial.write('\n');
      delay(10);`

salopette, 2018-03-04:
How and where can I change the baudrate?

KPK75, 2018-03-04:
In sonoff.ino, GpioInit procedure, look for Lctech and baudrate=

KPK75, 2018-03-04:
@mischkasmith adding delays can have negative impact on the rest of code. Is 20ms minimum value needed to work properly?

mischkasmith, 2018-03-04:
@KPK75 With 10ms the second relay did not switch sometimes. Since the increase to 20 ms it switch every time.

KPK75, 2018-03-04:
@mischkasmith I forgot serial.flush() command. Could you test it now ? With and without delay() ?

mischkasmith, 2018-03-05:
@KPK75 Have try this before, this works not. Is there an variable for the button 1 and 2 which can use to get along without the for loop?

KPK75, 2018-03-05:
No. It's one variable. Also PowerOnState could triger all relays so we have to leave delay. Probably STM is not fast enough ( time to transmit one command is below 1ms ).

rt400, 2018-05-01:
hi @KPK75 can you plz upload a bin file for lctech v2 ?
i try to compile with arduino all good but after i flash it into esp-01 its seems no wifi at all..
maybe i did somthing wrong ? the bin file got 500KB but my esp-01 have 1MB..
best regards...yuval

highground88, 2018-06-04:
@KPK75  Thank you very much for your efforts - I easily just flashed 2 of the single relay devices, with no changes to your code (just changed module in GUI). Any chance this will be put into the main Tasmota git soon? I just tried SonWeb, and the auto upgrade, which ditched your changes unfortunately.

Of note... apparently I have 5v versions - that I've been running at 12v for some time now, and during the programming. Oops.

KPK75, 2018-06-06:
Ask Theo :) PR is ready

hasenradball, 2018-06-20:
Hello togehter,

I have also flashed my ESP8266 with tasmota to be able to switch my relais by sending a serial message with a baurate of 9600.
But how must I configure the tasmota to be able to send this messages?

With my arduino solution I am able to do this by serial write.

KPK75, 2018-06-21:
Assuming that your sequence for on/off is A00101A2 / A00100A1 (for Relay1) 
you need to define new template 
`  { "LCTECH 1Relay",     // LC Technology WiFi 1 Relay`
`     0,`
`     GPIO_TXD,          // GPIO01 Relay control`
`     0,`
`     GPIO_RXD,          // GPIO03 Relay control`
`     0,                 `
`     0,`
`     0, 0, 0, 0, 0, 0,  // Flash connection`
`     0,`
`     0,                `
`     0,                `
`     0, 0, 0`
`  },`

and add some code to function:
SetDevicePower() just below SONOFF_DUAL
`
else if ((LCTECH1 == Settings.module) || (LCTECH2 == Settings.module)) {`
`    for (byte i = 0; i < devices_present; i++) `
`      if ( i > 0 ) delay( 20 );`
`          state = rpower &1;`
`      Serial.write(0xA0);`
`      Serial.write(i+1);`
`      Serial.write(state);`
`      Serial.write(0xA0+i+1+state);`
`      Serial.flush();`
`      rpower >>= 1;   `
`    }`
`  }  `

and also in function void GpioInit() : 

`  else if (LCTECH1 == Settings.module) {`
`    Settings.flag.mqtt_serial = 0;`
`    devices_present = 1;`
`    baudrate = 9600;`
`  }`
`  else if (LCTECH2 == Settings.module) {`
`    Settings.flag.mqtt_serial = 0;`
`    devices_present = 2;`
`    baudrate = 115200;`
`  }`

hasenradball, 2018-06-21:
Hi thanks for your reply,

Is there actually a bin file available with this function, or is it planned to make a PR to the tasmota source?

I am actually not so firm with building tasmota.

Best regards

Mit freundlichen Grüßen

Frank Häfele

KPK75, 2018-06-21:
1) No
2) Ask Theo

hasenradball, 2018-08-01:
Hi,
trank you for Thees Information.
Is the 6.1.1.c already available.
In the Release I See only 6.1.1 Version.

KPK75, 2018-08-01:
Development branch

mattpurnell, 2018-08-04:
Hi @KPK75 - where can we get more info around this "MQTT raw data" approach? I also have several LC TECH Relay modules that I cannot control yet..

🥂

hasenradball, 2018-08-04:
Hi This command

"\xA0\x01\x01\xA2"

Dors only work with one relais. The hex meassage does depend on the number of relais and the state.

hasenradball, 2018-08-04:
You can see this message build routine some comments above in this chat.

ascillato2, 2018-08-26:
Hi,

Let's gather all LC Wifi Relay config issues in #1282

closed 2018-08-26 by ascillato2