arendst/Tasmota #1715: Sonoff won't connect to WiFi after router reboot:

shayshtern, 2018-01-26:
some of them connected some not.
any reason why?

SodoffBaldrick, 2018-01-26:
What status do the LEDs show? What flashing pattern?

shayshtern, 2018-01-26:
sonoff touch, now its avalliable I can find it on my Phone and I set my WiFi SSID details , but after that - its wont show up on my router

SodoffBaldrick, 2018-01-26:
Ah, I thought you were saying that they were working away fine then when you rebooted them they suddenly stopped connecting to your router.

Double and triple check the SSID and password.
Consider flashing using PlatformIO and including the WiFi details in the compiled firmware so you can just skip the manual WiFi join stage.

shayshtern, 2018-01-26:
Its an installed sonoff touch, I prefer not get it out from the wall.

shayshtern, 2018-01-26:
My WiFi called - "Shte Wi-Fi"
after I set it its reboot - but still available on my Phone but AP1 SSId says "Shte"

reloxx13, 2018-01-26:
> Its an installed sonoff touch, I prefer not get it out from the wall.

thats why you should always build the firmware manually with the right wifi config set in the user_config.h
---

seems it does not like your whitespace. and special chars in the ssid, maybe in pw, too

ionciubotaru, 2018-01-27:
Maybe is this bug https://github.com/esp8266/Arduino/issues/4078

closed 2018-06-11 by stale[bot]

gammerboy, 2020-05-25:
Having same issue where the light stops connecting to router (when i restarted the router and this issue is repeating for me) and broadcasts its own ssid and also the light switches off for a second and turns back on once every few minutes.