arendst/Tasmota #1716: Detecting power switch pressing on Sonoff B1

BlackLotus, 2018-01-26:
The Sonoff B1 can be brought into flash mode by pressing the light switch 5 times with 2 seconds pause between presses.
I would like to configure events based on power changes, but couldn't yet figure out how to do this correctly.
Is there already a way implemented in Tsamota?

arendst, 2018-01-26:
Nope.

ionciubotaru, 2018-01-27:
There is no reason to implement this in Tasmota. You can do it in you HA software.
ESP has limited memory and programming space, so only basic features could be implemented.

BlackLotus, 2018-01-27:
How could I implement this in my home automation software?
I don't see how I can get this information over mqtt is there something obvious I'm missing?

ionciubotaru, 2018-01-27:
> I would like to configure events based on power changes ....

Every time the power status is changing Tasmota publish a mqtt message with current status ON or OFF

closed 2018-06-11 by stale[bot]