arendst/Tasmota #1724: Button1 on Wemos D1 create reset

fdmarder, 2018-01-27:
Hi,
Want to use a WEMOS D1 mini for my garage door automation.
The modul is programmed with tasmota 5.11.1.
I configure the modul as Modul type „18 WeMos D1 mini“,
the „D1 GPIO5“ as „Button1“ and „D5 GPIO14“ as „Relay1“.
After saveing this setup the modul makes a reset and switch back to a Modul Type
„01 Sonoff Basic“. 
With „D1 GPIO5“ as „Switch1“ there is no problem. 
What is the functional different between Button and Switch ?

ascillato, 2018-01-27:
Please, try to flash Tasmota v5.11.1g (**Wemos D1** is now **GENERIC**)

I test the config you specified and works with Tasmota v5.11.1g.

Button:
  - Press and release --> output ON
  - press and release again --> output OFF

Switch
   - Press --> output ON
   - release --> output OFF
     ( like a light switch of a house )

fdmarder, 2018-01-27:
Hi,
 I test it again with the 5.11.1g , but the same behavior. 
Typ: 18 Generic, „D1 GPIO5“ as „Button1“ and „D5 GPIO14“ as „Relay1“.
After saveing the modul makes a reset and switch back to a Modul Type „01 Sonoff Basic“.

ascillato, 2018-01-27:
check GPIO0. It is connected? because it has a [special behaviour](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage)

fdmarder, 2018-01-28:
no connection on GPIO0,  
the console shows "RestartReason":"Software/System restart"

hwarang098tkd, 2018-02-22:
same problem