arendst/Tasmota #1725: Connecting Tasmota to IBM cloud

sush0408, 2018-01-27:
I was trying to connect Tasmota to IBM IOT platform but the problem is IBM cloud MQTT address is quite long and get snipped off from last.

 > Actual host address: "XXXXXX.messaging.internetofthings.ibmcloud.com"  

after entering this into MQTT configuration the address gets wrong  
 "XXXXXX.messaging.internetofthings.ibmcloud.com"  which is due to 32 char limitation of host address in tasmota firmware.

Is there any workaround to get it working or if anyone has ever connected Tasmota to IBM cloud please help!!

Regards 
Sushil

ionciubotaru, 2018-01-28:
In `settings.h` try to change this line `  char          mqtt_host[33];` to a greater value.
Upload the new sketch, then send `reset 2` via serial or web console, then try to connect to IBM cloud.

sush0408, 2018-01-28:
Getting an Error in IBM Logs: 

> Closed connection from 14.139.194.12. The topic is not valid: iot-2/evt/status/fmt/json/LWT The topic does not match an allowed rule

yozmag, 2018-03-24:
I'm having the same issue, did you solved it?

lobradov, 2018-03-24:
So, that means you managed to connect but you have some other issue?

yozmag, 2018-03-26:
I just replaced the domain with the ipaddress, but the issue with the topic is very difficult IBM requires strict topic rules, like:
Event topic format: iot-2/evt/event_id/fmt/format_string
Where the event_id specifies the event name that is shown in Watson IoT Platform, and format_string is the format of the event, such as JSON. 
Message format: JSON
Watson IoT Platform supports several formats, such as JSON and text.
From https://github.com/ibm-watson-iot/device-arduino/blob/master/samples/ESP8266MqttSecure/ESP8266MqttSecure.ino
There is a an exemple String payload = "{\"d\":{\"myName\":\"ESP8266.Test5\",\"counter\":";
I don’t think this is doable with tasmota.

miklik72, 2018-04-16:
I have the same problem with long hostname for IBM IoT Watson. It have 45 characters. When I increased char array to 50, Tasmota was not running in ESP and restarted by watchdog.

  char          mqtt_host[50];             // 1E9

closed 2018-06-25 by stale[bot]