arendst/Tasmota #1728: Program Flash Size vs Flash Size

Frogmore42, 2018-01-27:
This might be a stupid question, but I did not see anything conclusive in my quick search.  Here is what my new S31 device says:

Flash Chip Id 		1458415
Flash Size		         4096kB
Program Flash Size	 1024kB
Program Size		  464kB
Free Program Space  536kB
Free Memory		    23kB

So, it looks like it has a 4MB flash.  Does that mean it will be able to easily flash a larger 500KB program.  For a 1MB flash size device the answer would, of course be no, at least not a >500KB to >500KB one.  But, for a device with a larger flash chip like this, it seems like it should be possible to put the new program in the extra space and then move/run it from there.  I have not looked at how the linker links things and how the loader loads things to know what the code does.  Right now it is not an issue, since I am still using an older version that is small enough.  I have resisted moving forward, since it seems there is some instability with some of the recent changes.  I am sure I will move forward, if for no other reason than full support for power monitoring of the S31.

ascillato, 2018-01-27:
Depending on your device, you can have 512kbits, 1024Kbits, 2048Kbits, 4Mbits, 8 or 16

then all that flash memory is splitted on program flash, OTA space for updates, SPIFFS (a file system space) and libraries (wifi etc)

following is the OTA (over the air) update process:

[image]

Frogmore42, 2018-01-28:
That picture doesn't have any sizes, but I searched and found that the esp8266 can only access 1MB at a time.  It looks like the answer to my question is, no it will not be easy to flash a program larger than 500KB.  However, it looks like there might be a solution with rboot [https://github.com/raburton/rboot](url)

This will allow up to the full 1MB for each image and still allow for space for persistent memory, but I am pretty sure it would not be really easy to support this.  For one, keeping the stuff stored in eeprom would have to be handled differently.

ascillato, 2018-01-28:
The picture don't have numbers because depends on YOUR DEVICE how much
memory it has. A nodemcu has 4Mbits so, if you are not going to use OTA and
SPIFFS you have 4Mbits of program memory. On arduino you can choose this
before flashing.
Also, you can change the memory chip up to 16Mbits.

https://github.com/raburton/rboot <http://url>

Frogmore42, 2018-01-28:
NodeMCU normally has 4MBytes (which is 32MBits) not 4MBits (which is 512KBytes).  But, from what I have read, the esp8266 can only address 1MByte of program memory at a time (so that is the maximum size of a program, but it is possible to access more flash for other data).  This article indicates it is possible (on a 4MByte part) to split up the memory to allow OTA of up to 1MByte programs: www.esp8266.com/viewtopic.php?p=72605

This article https://github.com/esp8266/arduino-esp8266fs-plugin/issues/14 makes it look like it might be possible without using rboot.  So, I might have to try that.  I have several boards that have 4MByte flash and it would make OTA much better for them, given the size of the code now.

I was surprised to see that my S31 has 4MBytes of flash, since many of my other Sonoffs only have the 1MByte of flash.

arendst, 2018-01-28:
@Frogmore42  you are right regarding the max 1M program size. Larger flash can only be used for data like spiffs or ota image.

Tasmota will only support the current version of OTA which, as you can read in Theos Tips, still provides OTA for images larger than 500k. Just use the 2 step approach.

ascillato, 2018-01-28:
Good to know! Thanks.

closed 2018-01-28 by Frogmore42