arendst/Tasmota #1729: how to change module setting

app41, 2018-01-27:
hello to all the tasmota community, this is a fantastic project
Need your help to understand

I compiled my sonoff package with 
#define MODULE SONOFF_4CHPRO        in sonoff.h
#define WIFI_IP_ADDRESS        "0.0.0.0"   in user_config.h
genral strategy I wonder should be I have a single build, I can use on all the sonoff devices, later when I switch them on the first time I can configure the proper module identity
The problem is I was not able to change module id. I tried through the web interface, no result
I tried also through the console, neither there any result
module setting seems to be locked and impossible to change

In this snapshot I'm running a sonoff basic
at startup time on the console:
00:00:00 Project sonoff222 Sonoff (Topic sonoff222, Fallback DVES_96E91D, GroupTopic sonoffs) Version 5.10.0g
00:00:00 WIF: Connecting to AP2 APP_NET in mode 11N as sonoff222-2333...
00:00:04 WIF: Connected
00:00:04 DNS: Initialized
00:00:04 HTP: Web server active on sonoff222-2333.local with IP address 192.168.10.108
00:00:06 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: letto4/sonoff222/LWT = Online (retained)
00:00:07 MQT: letto4/sonoff222/cmnd/POWER = 
00:00:07 MQT: letto4/sonoff222/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.10.0g","FallbackTopic":"DVES_96E91D","GroupTopic":"sonoffs"}
00:00:07 MQT: letto4/sonoff222/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff222-2333","IPAddress":"192.168.10.108"}
00:00:07 MQT: letto4/sonoff222/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: letto4/sonoff222/RESULT = {"POWER1":"OFF"}
00:00:08 MQT: letto4/sonoff222/POWER1 = OFF
00:00:08 MQT: letto4/sonoff222/RESULT = {"POWER2":"OFF"}
00:00:08 MQT: letto4/sonoff222/POWER2 = OFF
00:00:08 MQT: letto4/sonoff222/RESULT = {"POWER3":"OFF"}
00:00:08 MQT: letto4/sonoff222/POWER3 = OFF
00:00:08 MQT: letto4/sonoff222/RESULT = {"POWER4":"OFF"}
00:00:08 MQT: letto4/sonoff222/POWER4 = OFF
22:03:39 MQT: letto4/sonoff222/STATE = {"Time":"2018-01-27T22:03:39","Uptime":0,"Vcc":3.190,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":2,"SSId":"APP_NET","RSSI":86,"APMac":"xxxxxxxxxx"}}
22:03:47 CMD: module
22:03:47 MQT: letto4/sonoff222/RESULT = {"Module":"23 (Sonoff 4CH Pro)"}

after that in the console I write "module 1", it responds positively, it resets but when back it is still on 23/CH4PRO
what do I make wrong?

ionciubotaru, 2018-01-28:
try to change it via web interface

carryonrewardless, 2018-01-28:
My issue is similar but not identical. I am using Version 5.11.1g on a Sonoff POW but in the web interface it is not saving a change of configuration to the POW and still registers as a Basic module.
I could re-flash with #define MODULE SONOFF_POW in sonoff.h I guess but am I using the tool incorrectly or maybe a small bug? 
Thanks for the amazing effort in producing tasmota and the support from the community.

carryonrewardless, 2018-01-28:
Ok there is more to it for me. No configuration changes seem to be saved including telemetry interval and mqtt enable.
Would #define MQTT_USE 0 in user_config_override.h prevent subsequent changes? I assumed this would only apply for the initial flash/boot.

app41, 2018-01-28:
forgot to mention, version is 5.10.0g
@ionciubotaru: it is the first thing I tried, no effect
but mqtt configuration and ip setting work fine, tested on web interface

carryonrewardless, 2018-01-28:
Re-flashed with original, unchanged version of user_config_override.h. Set SSID & pwd & timezone only in user_config.h.  Config changes are now saved in web interface.

app41, 2018-01-29:
just made the same test
STEP 1
recompiled with  #define MODULE SONOFF_BASIC
Downloaded through 5pin port, no ota
It starts but he thinks still to be a CH4PRO, web interface is showing the 4 relays layout
Tried to modify the module value via web interface, it reset and crash, no anymore answer, no relay movement with local pushbutton, bricked
STEP 2
Changed also CFG_HOLDER number to force eeprom reformatting, total new build
Downloaded new firmware in the sonoff basic, now all is running
I can change any configuration via web interface

let me add that I downloaded the firmware several times on this device before having the problem
it seems something wrong in eeprom data management

carryonrewardless, 2018-01-30:
@app41 You can also use this command to clear the flash:
python esptool.py --port /dev/ttyUSB0 erase_flash
and then upload again.  Seems more reliable than CFG_HOLDER method.

closed 2018-06-15 by stale[bot]