arendst/Tasmota #1732: Sonoff T1 EU (2 ch) - does not work

wzbart, 2018-01-28:
Does anyone tried to flash EU version of T1? I followed the instruction from wiki and new firmware seems to successfully landed in T1. It's getting connected to my home network and I'm able to reach tasmota web server from T1. Unfortunately the issues starts from here. I'm yet able to change sonoff type in configuration but after that nothing is working as should. I can't really describe what is going on as I can't see any repeatable pattern. Sometime when I touch fist button nothing is happening and sometime second channel is changing it's state. Second touch panel is usually not working at all. Same if it comet to web front end. When I toggle the channel nothing is usually happening. In the end nothing is actually working and web server goes unreachable.  
From what I can see the board of T1 EU looks a bit different than UK version. I'm guessing that might be the issue as GPIO might be connected differently than in UK version. 
Dis anyone tried to flash EU version?

gajotnt, 2018-01-28:
Check if mqtt server is right.

wzbart, 2018-01-28:
MQTT would be next step for me. Thing is, T1 doesn't work itself. Touch panels does not work as they should. This is a firmware issue for sure.

gajotnt, 2018-01-28:
Check the logs...I bet it's trying to connect to a mqtt server and for what it seams you don't have one.

wzbart, 2018-01-28:
Actually I do have MQTT server running. It's absolutely fine considering I have flashed sonoff POW shortly before T1 and it woks great. All my other sonoffs are working fine too. I have got couple Touch models, few basic and TH16 .. all works just fine. I'm flashing all sonoffs with same tasmota version (5.10 I guess) and with same ardurino settings. The only thing that is changing is device (topic) name

Pajn, 2018-01-28:
I have a Sonoff T1 2ch EU version flashed with Tasmota (albeit not the latest version, but the one that was included in SonOTA which, if I remember correctly is the November release).
It works fine apart from the buttons being a bit unresponsive so I have to hold them down some time and/or press hard.

wzbart, 2018-01-29:
Eventually I have figured this out. I'm suspecting power drop glitches while testing after firmware change. I had sonoff connected to 3.3V from my programmer. Probably VCC or GND were getting lose causing sonoff to reboot. I have put it in to the wall and all seems to be fine so far.

pkaras3, 2018-01-30:
What type of switch did you choose in the configuration menu?
Does the two switches work in different MQTT context or there is a difference to distinguish which one was pressed in the MQTT command?

wzbart, 2018-01-31:
I have chosen T1 2CH. Here is how the MQTT comms looks like:
`09:13:52 MQT: stat/ST1_2/RESULT = {"POWER2":"OFF"}
09:13:52 MQT: stat/ST1_2/POWER2 = OFF
09:13:53 MQT: stat/ST1_2/RESULT = {"POWER1":"OFF"}
09:13:53 MQT: stat/ST1_2/POWER1 = OFF`

mattpurnell, 2018-02-09:
Hi - how you get that to work? My T1 2ch (two buttons) acts as if it only has 1 button; the second does nothing. Did you use a different firmware? 

P.S. I haven't been able to get much sense out of the webserver, even on models that are working fine. Are there some steps in the web server I need to take?

gajotnt, 2018-02-09:
You have to set it has a 2 channel T1 and it will work

wzbart, 2018-02-09:
> Hi - how you get that to work? My T1 2ch (two buttons) acts as if it only has 1 button; the second does nothing. Did you use a different firmware?

You need to set your sonoff as T1 2CH (exactly like gajotnt mentioned abowe). 
Type your sonoff's IP in web browser and on the sonoff's web server go to "Configuration" -> "Configure Module" and there from the list select "29 Sonoff T1 2CH" ... save and you're done

mattpurnell, 2018-02-10:
Thanks guys - I clearly need to get the webserver going. For some reason the webserver never displays right on the Tasmota devices I flash - https://github.com/arendst/Sonoff-Tasmota/issues/1839#issuecomment-364349754

I have been told you can configure the module via mqtt command - so I am currently looking for that info.

mattpurnell, 2018-02-22:
Hi all - just wanted to send an update; I couldn't get the webserver working at all. 

However it is dead easy sending an MQTT command to configure these devices. To paraphrase the instructions; sending the MQTT topic "cmnd/lightlngdk/Modules" will prompt a response from teh device listing all the available modules it can emulate. You look through the result and find the module you are  after - then send a the MQTT topic "cmnd/lightlngdk/Module" with the payload being the index number - 29 for the T1 2CH. Then the device restarts as that module. Dead easy.

closed 2018-06-04 by ascillato