arendst/Tasmota #1734: Upgrade firmware from my pc with commands

hwarang098tkd, 2018-01-28:
HI
I tried to find commands for upgrading sonoffs from my pc
otaulr to set the link? (C:\Users\Hwa Rang\Desktop\sonoff.bin)
and then upgrade 1? or upload 1?

curzon01, 2018-01-28:
You need an URL which is avalable via http. It does not work with a simple file located on your PC. 
If you want to upgrade via a local file you can install a web server on you PC which provides the file via a local url.

But do you think this is an issue? Whats about move to [discussion forum](https://groups.google.com/forum/#!forum/sonoffusers)

hwarang098tkd, 2018-01-28:
ok i did it. i establish a web server in my pc (win 10).
I noticed that if i run this url to my browser i can download the file .
localhost/sonoff.bin
i set it with the command otaurl localhost/sonoff.bin and then upgrade 1 but 

00:02:50 MQT: stat/sonoff_t_8/RESULT = {"OtaUrl":"http://localhost/sonoff.bin"}
00:02:56 CMD: upgrade 1
00:02:56 MQT: stat/sonoff_t_8/RESULT = {"Upgrade":"Version 5.8.0 from http://localhost/sonoff.bin"}
00:02:58 HTP: Web server stopped
00:03:06 HTP: Web server active on sonoff_t_8-5539.local with IP address 192.168.1.9
and then restart!!!!!!!!!!!

curzon01, 2018-01-29:
localhost is a address only visible by your own pc, localhost is not an address which is accessible from other network devices within your local network.
You need be able to access the file via an address within your network: e.g. by IP or by PC network name. Try access your file from your pc via http://192.168.1.x/sonoff.bin  (x is the IP of your PC). If this works, try on Sonoff device using this url.

hwarang098tkd, 2018-01-29:
ok i did it. it was hard for me but i did it . thnx

closed 2018-01-29 by hwarang098tkd