arendst/Tasmota #1736: Advice/Help/Guidance with RF Bridge & 433 Mhz devices.

erew123, 2018-01-28:
Im completely new to working with both the RF Bridge and 433Mhz devices. My bridge is flashed and working fine. Ive read the wiki and Ive read as many posts on here as I can, but Im not able to put all the information together in my head properly, or Ive not found a suitable article yet. If I can figure all this out, Im happy to update the Wiki on here to make it clearer for others. 

Its really a question of, what am I supposed to do with the following devices (do they need to be paired? Not paired? etc). I have bought:

433Mhz Door Sensors (Support Open/Closed and Tamper notification)
433Mhz PIR sensor (Supports No Motion/Motion and Tamper)
433Mhz Keychain remote control (It has 4 buttons on it).

So, for example, I have 3 x Door sensors as above. I put the battery in them and the RF bridge just sees them and sends out a MQTT signal, which is great...... but are you meant to pair them? Would there be a benefit to doing that (security? differentiating the MQTT, which I can already do simply by the DATA code in the JSON MQTT)?

1) So, what do you need to pair and not pair out of these kind of devices?
2) What is the point of pairing devices (what are the benefits)?
3) Can you give an example of the commands to pair a device e.g. the 4 button remote?
4) In the interface of the Bridge, there are buttons 1 to 16. If I have a 4 button remote, do all 4 physical buttons on the remote, get programmed into button 1 (device 1?) of the Bridge webpage? OR would you program button 1 to button 1, 2 to 2, 3 to 3 etc.....

Any guidance/links/insights anyone can provide on setting up these types of devices would be highly appreciated!

Thanks

erew123, 2018-01-30:
As no-one has replied yet, here is my so far guess/understanding.

**1. So, what do you need to pair and not pair out of these kind of devices?**
The Keychain remote controls or anything that uses Rolling Codes, so that the Bridge can identify/calculate future Rolling codes being sent from the Keychain remote (or other device that uses rolling codes). https://en.wikipedia.org/wiki/Rolling_code

**2. What is the point of pairing devices (what are the benefits)? Is it Security or simply that you can emulate your device sending a code by pressing the buttons in the Tasmota interface?**
Technically security to ensure that approved rolling code devices are the ones sending the code and to calculate the future codes they may send (as above in 1)

**3. Can you give an example of the commands to pair a device e.g. the 4 button remote?**
Im assuming just the "rfKey" commands at the console is the only way to do it. https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#sonoff-rf-bridge-433

So for example, to program Tasmota interface, virtual button 7 to a button on the remote, you would in the Tasmota console type **rfkey7 2** with the 7 specifying the virtual button and the 2 being the command to tell the bridge to go into leaning mode for that virtual button. Once you have typed that, you would press your button on your remote. (though I still dont understand the High/Low parameters).

**4. In the interface of the Bridge, there are buttons 1 to 16. If I have a 4 button remote, do all 4 physical buttons on the remote, get programmed into button 1 (device 1?) of the Bridge webpage? OR would you program button 1 to button 1, 2 to 2, 3 to 3 etc.....**
It appears to be a 1 to 1 relationship.... so a 4 button remote, if you programmed all buttons on that remote to do something, you would have to **rfkey** 4 of the virtual buttons in the web interface for the bridge.

**If anyone can confirm or correct any of the above, it would be greatly appreciated.**

tyjtyj, 2018-02-08:
@erew123, off topic. I dont know how to contact you.

I got sonoff rf bridge but I am not able to flash tasmota to the rf bridge. Keep getting error md5 sum does not match using esptool. May I know how to flash your rf bridge ?

erew123, 2018-02-09:
Please see this Ive just written https://github.com/arendst/Sonoff-Tasmota/issues/1841

tyjtyj, 2018-02-10:
Got it... Thanks..... Found issue on usb to ttl  causing the flash to fail...

Jason2866, 2018-06-19:
@erew123 
Can you give me a link for the 433Mhz Door Sensor (Support Open/Closed and Tamper notification)?
I just found Sensors who sends a code when opening
Thx

erew123, 2018-06-22:
The types of sensors I used were Kerui D026. You can find them all over the internet for a variety of prices, though these are relatively cheap for the features they offer.

https://www.banggood.com/KERUI-D026-433MHz-Wireless-Door-Window-Magnetic-Sensor-Dedector-Alarm-System-p-1309990.html?

closed 2018-06-22 by erew123

Jason2866, 2018-06-22:
Thx, if i understand the description right it only sends a code when opened.
No code sended when contact is openend and is closed

erew123, 2018-06-22:
These models send 4 separate codes for:

- Open
- Closed
- Tamper
- Low battery

I have them set up within a home automation system, which I built these in as an alarm portion.... and I get all the above status fed over to the bridge.

You can find examples of the sort of code/responses into the bridge from these devices, where I discussed here with people about how they work:

https://community.openhab.org/t/solved-mqtt-jsonpath-transformation-strange-results-in-basic-ui/39336

They definitely send those codes and you will get a status code for a door being opened and a door being closed (along with tamper and low battery).

erew123, 2018-06-22:
The D026's are also discussed midway through that post.

Jason2866, 2018-06-22:
Great! I will order some
THX

ascillato, 2018-06-22:
Thanks :+1:

ripvega, 2019-01-31:
> These models send 4 separate codes for:
>
> * Open
> * Closed
> * Tamper
> * Low battery
>
> I have them set up within a home automation system, which I built these in as an alarm portion.... and I get all the [...]

The open, closed and tamper codes are easily found. How did you determine the battery code and what is it's trigger level?

peret2000, 2019-05-14:
After lots of readings, I do not have any clear idea regarding rolling codes with Tasmota firmware in Sonoff RF Bridge.
The 'native' device (with its original firmware) seems not to allow rolling codes (that is clear); but does Tasmota allow them?
Not clear, and this is the only thread where, thanks to @erew123, I have found some clue: you say that pairing devices will allow it, won't it?
Has anybody put it to work with rolling codes? Maybe it is an obvious questions, but I am not able to find documentation about it.
I prefer not to invest in a device if finally it will be useless; all my remote devices use rolling codes.
Thanks a lot!

Jason2866, 2019-05-14:
Rolling codes are NOT supported. The hardware is too limited for.

peret2000, 2019-05-15:
Thanks a lot, @Jason2866!

lbriet, 2019-08-24:
@erew123 - same question as @peret2000 pretty much. All info I can find suggests that the RF Bridge is incompatible/can’t be used for rolling code RF devices (exactly as Jason says above).

You mention you used the keychain remote to pair with rolling code devices specifically. Can you elaborate on that a bit more? Like many others (I think), trying to figure out if (maybe with some workarounds) Sonoff RF Bridge can be used to automate garage door and house alarm.

yabuts, 2020-02-05:
So anyone has an idea what can be used for rolling code instead of sonoff bridge, I dont want to use a none rolling code to open my doors as it is not safe and can easily be duplicated by anyone.

mcaden, 2020-02-05:
@erew123 @ripvega and anybody else trying to decode the d026's

I've been decoding my d026's. The data I'm getting on tasmota via Sonoff RF bridge is 6 hex digits, with the last one being the identifier. It seems to be:

- `7` - Close
- `B` - Tamper
- `E` - Open
- `F` - Low Battery

To check the low battery one I took the battery out and using some wires looped in a lower-voltage batteries. (in my n00bness knew this probably wasnt a good idea but figured I could afford to write off one sensor as an experiment if I ruined it). It gave the same `F` several times so I assume it's right. I'll edit this post later as I finish mapping out all my sensors if this scheme doesn't seem to hold up.

erew123, 2020-02-06:
@mcaden - From my own decoding, yes this is the rule format they use. At least, mine follow the same format. So Id assume that they all use the 

7 - Close
B - Tamper
E - Open
F - Low Battery

Its at least a good guide.

@yabuts - No ideas. I think its a hardware chip thing, rather than a software thing. Ive tried alternate firmware on my bridge, but found the whole process frustrating overall and didnt actually get any additional features or rolling code support. Maybe someone else beyond myself has some ideas.

peterson1190, 2020-07-31:
> @mcaden - From my own decoding, yes this is the rule format they use. At least, mine follow the same format. So Id assume that they all use the
>
> 7 - Close
> B - Tamper
> E - Open
> F - Low Battery
>
> Its at [...]

Hello,

Wich Rfraw mode are you using to detect the code? 
I tried 1, 166, 167, 177 and only 177 gave me something but this is useless for me.
Wich .HEX are you using ?