arendst/Tasmota #1738: [Solved] Bug on GpioInit()

ascillato, 2018-01-28:
The actual code on the subroutine `GpioInit()` on _sonoff.ino_ uses `MAX_RELAYS` const as _PIN COUNT_ when checking the configuration for setting up the Outputs.
```
    for (byte i = 0; i < MAX_RELAYS; i++) {
      if (pin[GPIO_REL1 +i] < 99) {
        pinMode(pin[GPIO_REL1 +i], OUTPUT);
        devices_present++;
      }
    }
```
The bug shows up when you have a Relay on a GPIO# with number more than 8 than the Relay1 (MAX_RELAYS const).

**Example:**
Relay 1 on GPIO2, so `GPIO_REL1 = 2`
Relay 2 on GPIO12 so `if (pin[GPIO_REL1 +i] < 99)` will not arrive to GPIO12, as GPIO_REL1 + 8 = 10 

**Other example:**
Using Generic Module
`GPIO_REL1` is going to be 0
so, if you have a Relay on GPIO12 is not going to work and is not going to be on HomeAssistant Discovery neither.

So, the proposed code to solve the bug is:
```
    for (byte i = 0; i < MAX_GPIO_PIN; i++) {
      if (pin[GPIO_REL1 +i] < 99) {
        pinMode(pin[GPIO_REL1 +i], OUTPUT);
        devices_present++;
      }
    }
```

ascillato, 2018-01-30:
arendst:
> MAX_RELAYS is defined as 8.
> 
> This is the max index value for GPIO_REL1 to GPIO_REL8. Your suggestion will overflow max relays.

Ok, so the approach should be another as the Bug still exists.

If you select generic module and you put a relay on GPIO14 for example, it is not going to work, as explained

I'm reopening #1738 for further analysis.

arendst, 2018-01-30:
If this was a bug then how do you explain why a Sonoff 4CH is working fine with relays connected to GPIO4, 5, 12 and 15?

Your assumptions are wrong. The pin array itself contains the pin number. It's not the index of the pin array.

ascillato, 2018-01-30:
You are right! That part of the code is right.

Sorry!

Was a bug in another part of my own code.

Thanks for the response. I have learnt a lot of how the Tasmota Code works.

Closing #1738 and #1739 (Please, dismiss both)

closed 2018-01-30 by ascillato