arendst/Tasmota #1741: On Sonoff B1 with Version 5.11.1g SetOption20 i got {"Command":"Unknown"}

store2704, 2018-01-29:
Hi,

i upraded to version 5.11.1g to verify the adjust Dimmer without turning on device funktion. In the release notes i read that option 20 is used. But when i type at the console the SetOption20 command i got the following error message? Is there something missing to activate Option 20?
08:57:58 CMD: SetOption20
08:57:58 MQT: /Sonoff/Schlafen/Sz.LampeSchrank/stat/RESULT = {"Command":"Unknown"}

best regadrs

Stefan

arendst, 2018-01-29:
Bug. Quick fix is change line 1002 in sonoff.ino from:
```
    else if ((CMND_SETOPTION == command_code) && ((index >= 0) && (index <= 19)) || ((index > 31) && (index <= P_MAX_PARAM8 +31))) {
```
in to
```
    else if ((CMND_SETOPTION == command_code) && ((index >= 0) && (index <= 20)) || ((index > 31) && (index <= P_MAX_PARAM8 +31))) {
```

store2704, 2018-01-29:
Change the line as suggested and it works. The test if the bulb stays off i will test later at home. Thanks for the quick support

closed 2018-01-30 by arendst