arendst/Tasmota #1742: External button

Kiwifranky, 2018-01-29:
HI,

I want to connect a free GPIO to a button and on a push of this button I want to receive a MQTT message. The button should not change the relais, but only be another digital input which is broadcast via MQTT. I did not find the right commands but maybe there exists something like this already in TASMOTA.

Your help is appreciated

Frank

Kiwifranky, 2018-01-29:
Found the solution...

Setup one UserGPIO as Relay2 and an other as Switch2. Now I get a MQTT message when the state of the input changes...

closed 2018-01-30 by arendst

gertmenke, 2018-05-05:
Is this also possible without sacrificing a UserGPIO for a virtual relay?

Kiwifranky, 2018-05-06:
I can't remember exactly, but I think the software will only activate the mechanism for relay/button2 when both are found... Try it...