arendst/Tasmota #1743: Problem with more than 100 WS2812

Giorgiofox, 2018-01-29:
Hello, 
I'm using an ESP-01 with 300 WS2812 on Gpio2

If I try to turn on at 100% of brightness these leds, all led begin to blink and ESP-01 begin to reset itself in loop. 

I've tried to reduce leds number, it happens also with 200 leds, but with only 100 leds, it works perfectly.

Please note that another ESP-01 with another (very simple) firmware, works well and is able to turn all at 100% the leds with white color so I think this is not a power related problem (I'm using a 5A 5V PSU)

What can I do/check?

thanks
Giorgio

ddamron, 2018-01-29:
This looks to me like you are running out of RAM.
For each LED, you need at least 3 bytes of RAM - enough to store the RGB values.  
Your simple firmware may work because it doesn't require the RAM overhead of this project.

Giorgiofox, 2018-01-29:
This is interesting. What I can do to solve?
I need only MQTT and WS2812

altelch, 2018-01-29:
Hi,

besides the RAM issue you should really use a stronger PSU. 5A is far 
from enought. A single type A 2812 needs about 32mA for white and type B 
about 52mA so you need 9.6A for 300x Type A or 15.6A for 300x Type B 
what both is a lot more than your 5A you have.

Greetings,

       Heiko.

ddamron, 2018-01-29:
I'm a beginner with this project, but not a beginner with WS2812's.
I would suggest to disable everything you don't need, HTTP, SSL (if enabled)..
Also consider using multiple devices?  put some leds on one and some on the other?
Try to free up as much RAM as you can..

Giorgiofox, 2018-01-29:
@altelch I agree with you, but is strange that with another firmware I can drive all 300 leds with no problem. I will try do increase memory & change PSU

Giorgiofox, 2018-01-29:
Increasing RAM (I've abut 26k free) not solve the problem. I have ordered a new 15A PSU, let's see

altelch, 2018-01-29:
Hi,

how long did you run the LEDs with your PSU driving them 0xFFFFFF? Is 
the ESP8266 powered by the same PSU? Maybe it's passing out when the PSU 
starts to drop in current. Had this effect when driving a shorter strip 
with a much to weak PSU.

Greetings,

       Altelch.

handsomejackuk, 2018-02-07:
Slightly off topic but i have 2 ws2812 LEDs attached to my sonoff from what i understand 50ma per pixel so 300 ma all LEDs on this is working off sonoff basic 5 v internal supply... What sort of current can the sonoff internal 5v supply provide ?

Giorgiofox, 2018-02-07:
Arrived today a 20A PSU.
Solved but still a mystery why a simple firmware can drive 300 leds.

closed 2018-02-07 by Giorgiofox