arendst/Tasmota #1745: DHT11 on my sonoff basic not show temperture

fgvicente, 2018-01-29:
I have connected a dht11 sensor in the gpio14 configure in modules the gpio but it does not show the values of the sensor

[image]

[image]

[image]

I do not know what can be wrong. thanks in advance

gtortone, 2018-01-29:
do you have a 4.7kohm resistor on DHT11 between 3.3V and DATA pin ?
(https://i0.wp.com/www.electrodummies.net/wp-content/uploads/2017/02/dht11-library-and-schematic-1.png)

fgvicente, 2018-01-29:
mmm no. I will add and I will retest. thanks for response

fgvicente, 2018-01-29:
already connect the resistance and neither does it. still does not show me the temperature. I even probe with the module of th but nothing.

[image]

[image]
 
23:18:37 CMD: sensor
23:18:37 MQT: stat/home/baño/extractor/RESULT = {"Command":"Unknown"}

It's like there's no sensor. how can I test it

gtortone, 2018-01-30:
I see you are using a quite old release. Try to upgrade it because there are some DHT11 issues solved in releases next to your...

lobradov, 2018-01-30:
...also, not sure Tasmota will accept non-English characters in URL (ñ)... try replacing with regular n to see if it would be better.

//_ on phone

app41, 2018-01-30:
really sure about connections? 
The devil is in the detail...

**SONOFF BASIC - CONNECTOR**
1 (close to the pushbutton)  Vcc
2 
3 
4 GND
5 data 
10k resistor between Vcc and data

**DHT CONNECTOR**
1  Vcc
2 data
3 
4 GND

by the way, I made a mistake in soldering a DHT22 last week, resistor was not connected to data
It run also without the resistor, maybe because of internal pull up in ESP

app41, 2018-01-30:
try also to recompile the whole package changing the eeprom validation number
(#define CFG_HOLDER in user_config.h)
and then firmware again your sonoff with the new image
should you have something dirty in your configuration, you can clear it starting from a fresh eeprom reformatting
it is a 2-minute test...

fgvicente, 2018-01-30:
im updated but still not working

[image]

Jason2866, 2018-01-30:
Maybe the sensor is defect?!

gtortone, 2018-01-30:
can you try the same sensor on a different board ? (e.g. Arduino, something else...)

gtortone, 2018-01-30:
if you connect a simple led on GPIO14 does it work ?

app41, 2018-01-30:
As last solution, try to cut/eliminate the resistor. I had similar experience few time ago, I don’t know why (maybe the sensor?) but it worked

fgvicente, 2018-01-31:
I already solved the problem. Review everything you do to me. the problem was bad welding in the sonoff. Thanks to everyone for their contributions

closed 2018-01-31 by fgvicente