arendst/Tasmota #1746: DS18B20 sample rate ?

myozone, 2018-01-29:
I'm plotting temperature from a DS18B20. I'm polling cmnd/NAME/status every second and plotting from stat/sonoff2/STATUS10  via JSON and a function
msg.payload = msg.payload.StatusSNS.DS18B20.Temperature;
return msg; 

However, this appears to give an updated temperature about every ~5 minutes, this is odd seeing I'am holding and releasing the DS18B20 every second and waving in the air.

What I'm I doing wrong ?
Dave.

myozone, 2018-01-29:
Ah ha, TelePeriod 10 fixed it. Is the above the correct way to read a DS18B20 ?

arendst, 2018-01-30:
You should use the output of a teleperiod message as you found out ;-)

myozone, 2018-01-30:
Ah, Thanks Theo. It just didn't seem right to me :) I'm still very new to all this. I'm super impressed with your firmware ! 
Thanks again,
Dave.

closed 2018-01-30 by arendst