arendst/Tasmota #1749: Sonoff cannot switch to other access point with same SSID?

hansies, 2018-01-30:
Hi there! I got my new Sonoff Pow working with domoticz.

All is looking good when I use it in my home, but when I want to connect it to the washingmachine in the garage I can't access the Tasmota gui (via ip-address in browser) anymore. When I bring it back into my home, it is available again.

It looks like this is because my access point I use in my home has no reach to the garage. In my garage I got WiFi via another access point with the same SSID but on another channel (6).

As far as I can see the Sonoff is sending data to domoticz, but the gui is not accessible.

Is this something that sounds familiar to anyone?

JF62, 2018-01-30:
Definitely not. I have a total of 5 (five) access points (all same SSID mostly all different channel) - and 13 different Sonoff and each of them connects to the best (nearest) access point.

JF62, 2018-01-30:
They switch by themself. Make sure you have wifi_retry configured

hansies, 2018-01-30:
Ok thnx, I will check it when I got it in my home again.
It's sending data from the garage now, so I got what I want. Just can't access the gui...

closed 2018-01-30 by hansies