arendst/Tasmota #1751: Change from no web + mqtts back to web server and mqtt

swampdogmash, 2018-01-30:
I originally thought i'd increase security by using no web server and implementing MQTTS instead. I'd like to swap back now to use the web server but some of the devices are difficult to access for serial flashing.

I been sending MQTT messages to them to update the OTA URL and can confirm with a "/otaurl ?" command that it has updated - I thought uploading a new bin file would overwrite my options but no success so is it possible?

I have been trying to do this using OTA, is it even possible or will i have to serial flash the devices?

mscreations, 2018-01-31:
You may want to issue the "reset 2" command to erase the flash, reset to user_config.h settings and then restart.

reloxx13, 2018-01-31:
yup, if u use the same CFG_HOLDER in teh user_config.h, the new config does not get loaded.

reset 2 might help, but this resets everything.

seems no way around.

swampdogmash, 2018-02-01:
Interestingly the existing code was supplied my myself directly from serial... the code I Then loaded was it's from sonoff.maddox.co.uk so I can't expect the cfg_holder was the same... it's statistically quite improbable. Curious one. Any other suggestions as I've tried these tna

swampdogmash, 2018-02-11:
Not sure it is  config issue.. having the web server OR the MQTTS is presumably a compiled in thing in the binary... I am quite new to this so not really sure why a completely different binary from a different source would not change the setup... Unless it didn't apply the new firmware of course....

swampdogmash, 2018-03-10:
Closing as have raised a separate issue with more detail and broader scope.

closed 2018-03-10 by swampdogmash