arendst/Tasmota #1752: mdns/Avahi

kskenyon, 2018-01-30:
I can't contact any of my devices using name.local 
.local addresses work fine for my other servers.
IP address works fine, too.

5.11.1f with ds18x20 

This used to work a few releases ago.

Kevin

smadds, 2018-02-01:
Works for me on latest release (5.11.1) - as long as I remember to add the mac numbers to the host name (e.g. office-leds-3249.local). 

btw, I had to give up on latest Virgin Media Hub as it didn't support .local dns resolution

ascillato, 2018-02-01:
If you use any Apple devices should work.

If you use windows you need to install the mDNS drivers. The easiest way is to install iTunes on windows and without restarting should work.

If you use any linux yoy have to install the mDNS Responder library

If you use Android you should install a mDNS app, but Will only work on the app. Is not going to work on Chrome

kskenyon, 2018-02-05:
I still have this problem.  mdns/Avahi/Bonjour works on Safari for any of my configured devices except Tasmota.  The config web page shows that mdns is active on Tasmota but it does not show up in any Bonjour Browser (MacOS 10.13).  It doesn't work in Safari, Chrome or Firefox.  It seems that something isn't right in Tasmota to work with these Mac devices and browsers.  It used to work until a month or so ago.  I use DHCP so my IP's change once in a while.  "sonoff-MAC".local would work so much better.  That is how I address my raspberry pi's ,etc.  No problems.

ocombita, 2018-08-17:
Accessing web interface in the form of <hostname>.local works pretty good in 6.1.1 and 5.12.0 from any browser, as well as ping request.

melyux, 2021-04-29:
Doesn't work anymore (if it ever did). It only "works" when connected to a router that adopts the devices' mDNS hostnames as their network-wide hostnames, which of course any browser can look up.

ascillato, 2021-04-29:
It is disabled by default. Please, check the docs on how to compile your own version with this feature enabled. Thanks.

melyux, 2021-04-29:
Do you mean Tasmota has to be compiled especially to make SetOption55 work? I didn’t see anything in the docs for that. Do you mean have it work at all, or have it work out of the box on first boot?

ascillato, 2021-04-29:
It is not included in any build.

Check the file that explains which features are included in which are not on every firmware type:

https://github.com/arendst/Tasmota/blob/development/BUILDS.md

Also, you can check in the configuration file (my_user_config.h) that the feature is not included and also the setoption55 is OFF.

https://github.com/arendst/Tasmota/blob/7a11a7b3a2fc8d6985eec206fd0a48794a4a141c/tasmota/my_user_config.h#L444-L447

https://github.com/arendst/Tasmota/blob/7a11a7b3a2fc8d6985eec206fd0a48794a4a141c/tasmota/my_user_config.h#L231

melyux, 2021-04-29:
Compiled it with the web discovery enabled using the Gitpod option and got a nice firmware with it enabled. Thanks! I didn’t know they even took it out of the compilation, thought it was just a by default disabled option.

Had to flash back to minimal firmware and then flash from that, didn’t accept it (buffer error) otherwise. Now it’s working!