arendst/Tasmota #1753: how can I modify the program to notify the status of a sonof every given time

vivi202, 2018-01-30:
sorry for my bad english thanks in advice

smadds, 2018-02-01:
From the web interface: Configuration, Configure Logging, Telemetry period. Set to number of seconds between updates.

From MQTT use command teleperiod (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

smadds, 2018-02-01:
Why would you want to?

The point of MQTT is that devices publish updates **when something changes** - and that change is passed on to subscribers. It saves endless polling traffic on the network. So, just subscribe to a topic and wait for an update.

To check the device is still alive, you can define the logging period (mqtt command teleperiod or from the web interface - Configuration, Configure logging, Telemetry period) and subscribe to tele messages.

closed 2018-06-16 by stale[bot]