arendst/Tasmota #1755: Sonoff Pow - energy usage between previous message and now

pjanco, 2018-01-30:
Hi,
I have connected 3W load, telemetry time period is 5 minutes.

"Period" energy value (energy usage between previous message and now) in MQTT message should be 0.3Watthour. But it is only zero.

Problem is, this "energy" variable is printed without decimal points. I think the 3 decimal points will be better for this purpose. Problem can be fixed by changing this line in file xdrv_03_energy.ino:
`dtostrfd(energy, Settings.flag2.wattage_resolution, energy_period_chr);`
into this:
`dtostrfd(energy, Settings.flag2.energy_resolution, energy_period_chr);`

arendst, 2018-01-30:
Why not change wattres to 3 to receive the same result?

pjanco, 2018-01-30:
"Settings.flag2.wattage_resolution" is common for "Power" and also for "Period". I wanted to keep resolution of "Power" in Watts without decimal points.

closed 2018-06-14 by stale[bot]