arendst/Tasmota #1757: Turning a Sonoff T1 EU 2CH into "3CH"

gajotnt, 2018-01-31:
SO i got myself a couple of sonoff T1 (EU version) 2 gang, and from what i can tell they have all the components installed to make a 3 sensor version (exept the relay).
Tryed configuring it has a T1 3 gang version but the middle sensor doesnt do anything, maybe the EU version has the middle one on another GPIO. 

Will try to MOD the FW to figure out the GPIO, but maybe someone already did this

Sonoff_template.h
The T1 3CH uses GPIO 00, 09 and 10, maybe in the EU version instead of GPIO10 its in another GPIO

```
  { "Sonoff T1 2CH",   // Sonoff T1 2CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     0,
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
  { "Sonoff T1 3CH",   // Sonoff T1 3CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Blue Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
```

[image]
[image]

Maskartpina, 2018-02-02:
Hi! Have you flashed tasmota into this switch?
I have 2 units and after flashing tasmota I can access the units and choose the T1 2CH unit in configuration menu but after a while it resets and I have the standart config with sonoff easy.
Can you help me?
Thanks
Rick

gajotnt, 2018-02-02:
What do you mean by "sonoff easy"?

Maskartpina, 2018-02-02:
I'm sorry. I mean Basic.
I think I've made some mistake with arduino ide parameters.
I've flashed tasmota without errors using arduino ide but it keeps rebooting and resetting configurations. 
I've previously flashed 10 sonoff basic an also s20 units that are working perfectly and fully integrated into home assistant. 
Thanks 
Rick

gajotnt, 2018-02-02:
Maybe you did something wrong, mine are working great.

rokdebevc, 2018-02-03:
@gajotnt i recently get one Sonoff T1 R2 EU and cannot figure out how to flash Tasmota on it. Can you give me some instruction. Thanks for help!
[image]

gajotnt, 2018-02-03:
Turn it on with the programer after its ON:
Touch Button 1 should then be held while the reset button (4) is pressed. This will cause the unit to reboot into flash mode.

To make sure this is confirmed on a serial console (serial monitor) (74880 baud) by the boot mode displaying (1,x) indicating that we are booted to the bootloader and not the flash.

marine1988, 2018-04-17:
I have tried 100 times to put it in flash mode no luck I allready have tasmota but i need to flash it again can you help me?

rokdebevc, 2018-04-17:
I solved my problem with switching RX and TX pin. :)