arendst/Tasmota #1758: Webpage has no styling

RolfKoenders, 2018-01-31:
Hello! 

Started playing yesterday with the Tasmota firmware for the first time. Everything went well and its working really nice! I can go to the ip address of the Sonoff and the UI shows up nicely and everything works as expected. 

Now the weird thing is that now (12 hours later) i try to open the webpage (over VPN) and it shows up without styling and with some javascript on the page. 

This is my raw response:
```
lementById('t').innerHTML='Restart in '+cn+' seconds';cn--;setTimeout(u,1000);}}function c(l){document.getElementById('s1').value=l.innerText||l.textContent;document.getElementById('p1').focus();}function la(p){var a='';if(la.arguments.length==1){a=p;clearTimeout(lt);}if(x!=null){x.abort();}x=new XMLHttpRequest();x.onreadystatechange=function(){if(x.readyState==4&&x.status==200){var s=x.responseText.replace(/{t}/g,"<table style='width:100%'>").replace(/{s}/g,"<tr><th>").replace(/{m}/g,"</th><td>").replace(/{e}/g,"</td></tr>").replace(/{c}/g,"%'><div style='text-align:center;font-weight:");document.getElementById('l1').innerHTML=s;}};x.open('GET','ay'+a,true);x.send();lt=setTimeout(la,2345);}function lb(p){la('?d='+p);}function lc(p){la('?t='+p);}</script><style>div,fieldset,input,select{padding:5px;font-size:1em;}input{width:100%;box-sizing:border-box;-webkit-box-sizing:border-box;-moz-box-sizing:border-box;}select{width:100%;}textarea{resize:none;width:98%;height:318px;padding:5px;overflow:auto;}body{text-align:center;font-family:verdana;}td{padding:0px;}button{border:0;border-radius:0.3rem;background-color:#1fa3ec;color:#fff;line-height:2.4rem;font-size:1.2rem;width:100%;-webkit-transition-duration:0.4s;transition-duration:0.4s;}button:hover{background-color/h3><h2>Sonoff</h2></div><div id='l1' name='l1'></div><table style='width:100%'><tr><td style='width:100'><button onclick='la("?o=1");'>Toggle</button></td></tr></table><br/><form action='cn' method='get'><button>Configuration</button></form><br/><form action='in' method='get'><button>Information</button></form><br/><form action='up' method='get'><button>Firmware Upgrade</button></form><br/><form action='cs' method='get'><button>Console</button></form><br/><form action='rb' method='get' onsubmit='return confirm("Confirm Restart");'><button>Restart</button></form><br/><div style='text-align:right;font-size:11px;'><hr/><a href='https://bit.ly/tasmota' target='_blank' style='color:#aaa;'>Sonoff-Tasmota 5.11.1h by Theo Arends</a></div></div></body></html>
``` 

The buttons are working though, i already tried restarting. 

I also tried to `curl` from a host inside my network but the same response is returned. 

Could this be due to the VPN connection? later today i can try it again from the same machine as i opened the webpage yesterday evening to see how it looks there.

Thanks!

reloxx13, 2018-01-31:
Which Firmware Version is running on the device?

RolfKoenders, 2018-01-31:
Sorry that i missed the relevant details: 
Its running on a: `Sonoff Basic`
And the line i uncomment in the `platformio.ini` file is:
```
env_default = sonoff
```

curzon01, 2018-01-31:
Thats not a firmware info - use "Status 2" on Console to get the firmware info.

If you have compiled it using lib 2.4.0 this behaviour might be normal. Use lib 2.3.0 instead.

If you do not know how to handle that I suggest for using already compiled binary (e.g. 5.11.1 or 5.10.0) form master branch: [https://github.com/arendst/Sonoff-Tasmota/releases](https://github.com/arendst/Sonoff-Tasmota/releases)

RolfKoenders, 2018-01-31:
I can check that later, the only version i can find currently is displayed in my raw response `Sonoff-Tasmota 5.11.1h`

curzon01, 2018-01-31:
What is the reason using a version from development branch (5.11.1h is not  from master)?
I really suggest using an already compiled binary (see above

if you really want to use a version from developer branch try compiling it with lib 2.3.0 and test again - for further info how to switch to another library use the very helpfull [Theo's Tasmota Tips](https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180103---tasmota-is-ready-for-esp8266arduino-version-240)

RolfKoenders, 2018-01-31:
Ah, now i see that 'development' is the default branch. I just cloned the repo, didn't paid that much attention to it to be honest. I will use the latest version. Still i find it a weird issue though.
Thanks for pointing it out @curzon01

curzon01, 2018-01-31:
so if issue is solved, can you close it?

arendst, 2018-02-01:
Looking at your raw response it seems you are missing packets of 256 bytes. One at the beginning and one somewhere half way thru. This is out of my control as it looks like a Webserver buffer or TCP  problem.

As noted here lib 2.3.0 is more stable although I use 2.4.0 with LwIP 1.4 and have not experienced packet loss my devices yet...

RolfKoenders, 2018-02-03:
As @curzon01 advised i took the latest release from the [release page](https://github.com/arendst/Sonoff-Tasmota/releases). Upgraded OTA (Which is a cool feature) and now it works as expected without any issues. Will close this issue. Thanks!

closed 2018-02-03 by RolfKoenders