arendst/Tasmota #1761: Add sliders for WS2812B and other LED-modules for RGB control at main menu

roesl, 2018-01-31:
Is it possible, to add sliders for RGB  control at main menu of tasmota modul in addition to the intensity slider ?

handsomejackuk, 2018-02-09:
yep this would be nice....

roesl, 2018-06-06:
Why is there no respond ???

ascillato, 2018-06-06:
Hi

Sorry by that. There were lots of issues.

Thanks for sharing your idea :+1:  I personally like it, mainly for fast testing when you connect the led strip, light, etc. Remember that also there are available the commands on the console.

By now, there are lots of things to work on and the most difficult thing is to keep the flash size and RAM as small as possible.

Tasmota is an alternative firmware for sonoff devices to be controlled by mqtt so as to be included in home automation. The original firmware can not be included in a home automation systems because it does not have mqtt. It just support (with internet connection) alexa and google api using Itead servers. Also remember that the original software does not have a webpage on the esp. You control your original sonoff using only the Itead's phone app.

By now, you can control the RGB easily using your home automation with pretty front-ends.

Also, it is the Yeti phone app supporting Tasmota that is getting better and better. (link in the wiki)

So, mainly my answer is: Like your idea, there are also some other options available for that, let's see what else can we do.

Thanks.

closed 2018-08-05 by stale[bot]