arendst/Tasmota #1762: Questions: How to Update from Tasmota to a newer Version...

freaktech, 2018-01-31:
.... without loosing my wifi settings or general config ?

When the switch is already installed.

With a OTA Update from the server i had a problem, it did not work.

I then used a offline file  and lost all settings and big problems to get a touch.switch in AP mode

i needed about 1 hour to fix this.

so what would be the best way step by step

thx
tom

smadds, 2018-02-01:
The settings should always be retained, Tom. I have done hundreds of OTA updates with generally no problems - as long as:

1. The devices were programmed properly to begin with. I use ESPTTool as detailed here: https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool. The dout mode of programming is very important.

1. You have good WiFi signal to the device. You's taking a risk if signal strength is less than 25%

OTA details here: https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade, including links to the binaries. Just save the appropriate URL into your device.

freaktech, 2018-02-01:
thank you, so you think after a wifi update OTA....

my WLAN settings should still be in the device?

to push the touch 4 times, its a pain in the ass to do this

smadds, 2018-02-01:
What state is the device in now? Can you check your router to see if it's been assigned an IP?

If it's going into AP mode, it means it can't connect to your Wifi. Catch it while it's transmitting (I use my phone to scan for SSID) and connect to the wifi, go to 192.168.4.1 and configure your WiFi credentials.

freaktech, 2018-02-01:
sorry

at the moment it is full working with i think FW 5.10

so when i flash now OTA  will it lost my wifi settings or should the stay?

smadds, 2018-02-01:
They should all stay

You can use `http://sonoff.maddox.co.uk/tasmota/sonoff.bin` as the OTA url if you are using the standard firmware and it will always be the latest released binary.

freaktech, 2018-02-01:
ok

will try it today

thx

freaktech, 2018-02-01:
so now i have the same situation like the last time

OTA update 
1-2 minutes nothing happened at the webinterface    and now i  think after a reboot its offline

at the router i cannot find a device called sonoff

so the update way is realy a problem   must uninstall a wood panel  to reset the sonoff to AP mode

or any other ideas?

ascillato, 2018-02-01:
if the firmware you use, have the same CFG_HOLDER in the user_config.h than the one you are updating, the old settings are not being reset.

So, if you are building the code you can check that. If not, may be that value is going to be differet making a reset of configuration. 

Also, you can set the wifi_manager to be as you need: wifi_manager,  wifi_retry,  etc

reloxx13, 2018-02-01:
**just dont use the bins on the releases page.**

download zip, change wifi settings and everything else in user_config.h, build and flash.
tip: read user_config.h from top to bottom and disable sensors/mqtt/dometicz if not needed.

better: use git, clone, change user_config, make a backup eg my_user_config.h, build and flash
new update: git revert user_config.h (to avoid conflicts), merge your settings from teh backup with ne new user_config.h, build & flash.

and never have problems again.

closed 2018-06-16 by stale[bot]