arendst/Tasmota #1763: Sonoff POW as meter only

danielschramm, 2018-01-31:
Hello,
I'm new to tasmota and just installed it on Sonoff Pow modules.
The energy measurement seems to be only enabled it the relays is switched on.

Since I only use them as power meter for devices which should always be on I soldered a bridge on mains.
With this firmware release I can only get readings when I power up the relays which is then 100% on and draws always some energy.

Were there any reasons why the measurement is disabled when the relays is off?

I intend to create a patch for this special use case. Is there interest to integrate it and should it be a config option?

Thanks for your opinions.

Daniel

CommodoreWhite, 2018-01-31:
Simply issue the command "poweronstate 1", the device will turn the relay "on" immediately after powers up and measure power until you turn it off.

Kind regards

danielschramm, 2018-01-31:
Hello,

of course that would be possible, but I intend to leave the relays off since it draws current causing unnecessary heat and power consumption.
Removing the relays completely and setting to one would also be a solution but I like the simple bridge which could easily be removed to undo the change.

Of course it's also easy to rewrite the firmware but then I would have a private branch which is not nice for the future.
So I just intended to start a chat If there were specific needs to disable the measurement during relays off? After understanding the reasons for the current implementation I might be able to supply a useful patch.

Daniel

mihalski, 2018-02-02:
Just how much power does a relay draw that this is an issue? I think the drop of solder might cost about as much as the saving you made.

Frogmore42, 2018-02-03:
A typical relay of the size in a POW uses about 150mW.  That is less than USD 0.20/year.  In 10 years that is about $2.  We have probably spent more time talking about it than it is worth for a single relay.  But if you multiply that by thousands, it might be worth implementing.

Someone else was asking for something similar before, but for a slightly different reason.

danielschramm, 2018-02-04:
As side effect of the higher current usage the thermal stress on the components is usually higher which reduces the lifetime of electrolytic capacitors. So leaving the relays off saves time and money over the next years. 
The bridge behind the relays  is needed in case of malfunction of the software. The tolerance level for a washing machine not working is very low ;-). Same for other systems I intend to monitor.

If there is already a similar issue we might add a config option to enable measurement always or only with relays on.

What was the intention behind switching off measurement when relays if off? I like to understand this design decision since it is done differently in other projects.

arendst, 2018-02-04:
Try the latest 5.11.1i with SetOption21 1...

danielschramm, 2018-02-05:
Hello,

thanks a lot, I just compiled and updated one module to the latest devel and it seems to work perfectly.
I also like the other idea of Master/Slave application very much.

Daniel

closed 2018-02-06 by arendst